openthread/openthread #15: Prerequisites for openthread?

rapport, 2016-05-11:
Thanks for open-sourcing this implementation. What are the pre-requisites to 'make' and run openthread from source code? i.e. for testing and getting things to run exactly as in your documentation what board/platform/system is recommended? 

I am trying out thread with android comissioning app, edison/brillo/weave, nxp and silabs and wondering how this works with these other implementations... Can you provide some pointers on getting started with this implementation and about the compatibility with other thread apps?

jwhui, 2016-05-11:
What's currently described in the documentation is running OpenThread in an emulated environment on Linux or Mac.  To build, you will need a gcc/clang toolchain.

OpenThread does not implement Thread Commissioning yet.

closed 2016-05-12 by jwhui

anujdutt9, 2016-08-31:
Hi. I want to test the implementation of openThread by running the openThread CLI example. As you said that it is supported on Linux and Mac, I tried running it on ubuntu but it does not recognizes the OS . Can you please tell which linux it supports for emulation and how to make it work.
Thanks : )

jwhui, 2016-08-31:
@anujdutt9, the continuous integration checks on [Travis CI](https://travis-ci.org/openthread/openthread) run on both Ubuntu and Mac.

From the travis logs on ubuntu:

> Operating System Details
Distributor ID:	Ubuntu
Description:	Ubuntu 12.04.5 LTS
Release:	12.04
Codename:	precise
Linux Version
3.13.0-92-generic