openthread/openthread #6: soc: netdataregister command should return Done

turon, 2016-05-11:
Running the netdataregister doesn't return the expected 'Done' response:

```
./soc --nodeid=1 -S
netdataregister
netdataregister
```

After running the command, other commands still work.  But if the command is run twice, the shell becomes unresponsive to all commands:

```
./soc --nodeid=1 -S
netdataregister
netdataregister
panid
panid
65535
Done
netdataregister
netdataregister
panid
panid
```

jwhui, 2016-05-11:
I submitted #16 to address the hang issue.

"Done" is only printed if the command completes successfully.  In your example, OpenThread was not started yet, so netdataregister is not successful.  Should we enhance the CLI to print "Error" with the error code instead?

turon, 2016-05-16:
Yes, "Error: <code or optional details>\r\n" would be a good format and deterministically parseable by the testing suite.

jwhui, 2016-05-24:
Fixed in #66.

closed 2016-05-24 by jwhui