openthread/openthread #9: Key agreement isn't demonstrated

briansmith, 2016-05-11:
I read the document saying that EC-JPAKE is used for key agreement but I can't find any use of JPAKE in the code. It would be great if the CLI could demonstrate the key agreement. (Also, it would be great if the *symmetric* key management that the CLI *does* implement was more clearly documented.)

jwhui, 2016-05-11:
OpenThread does not yet implement Thread Commissioning and, as a result, OpenThread does not currently use EC-JPAKE.

The current CLI doc for "masterkey" currently states setting and getting the "Master Key", which is the term used throughout the Thread Specification as the key material used to derive the MAC and MLE keys.  Can you suggest some improvements so I can better understand what you're looking for?

briansmith, 2016-05-12:
> OpenThread does not yet implement Thread Commissioning and, as a result, OpenThread does not currently use EC-JPAKE.

Is that coming soon? I am interested in implementing it a [*ring*](https://github.com/briansmith/ring)-based implementation that interops with openthread.

> The current CLI doc for "masterkey" currently states setting and getting the "Master Key", which is the term used throughout the Thread Specification as the key material used to derive the MAC and MLE keys. Can you suggest some improvements so I can better understand what you're looking for?

The "Getting Started" README doesn't mention was is needed to secure the network communication. Adding steps to set the master key on each node to the README seems like a good idea.

jwhui, 2016-05-12:
Thread Commissioning is on the roadmap, but we don't currently have a timeline.

Thread requires the network to operate with security enabled.  The example README currently relies on a hard-coded default configuration to hide those configuration details for now.  We could add a mention in the README to mention that.

jwhui, 2016-05-12:
I added a note with #20 .  Hope this addresses your comment.

closed 2016-05-12 by jwhui