openthread/openthread #31: NCP Implementation and Thread Host Controller Interface protocol

darconeous, 2016-05-14:
In addition to the "SoC/CLI" implementation that is already included, OpenThread needs a Network Control Processor (NCP) implementation to facilitate applications where the Host Controller is logically separate from the controller that implements the Thread stack (Which is the NCP). This task also necessitates the definition of the protocol that the Host Controller and the NCP use to communicate with each other.

I have developed a starting point for both of these, which are currently living in the development branch [`feature/ncp-spinel`](https://github.com/openthread/openthread/tree/feature/ncp-spinel). The initial name of the protocol is "[Spinel](https://en.wikipedia.org/wiki/Spinel)", but the long-term goal is to standardize the protocol to be the official Thread Host Controller Interface (or THCI).

A [draft of the Spinel protocol document](https://github.com/openthread/openthread/blob/feature/ncp-spinel/src/ncp/PROTOCOL.md#spinel-host-controller-interface) is included in this branch. Expect it to be updated regularly to improve readability and organization, as well as to incorporate received feedback and real-world experiences. At this point everything is subject to change, but as-is the protocol has been demonstrated to work fairly well.

The [NCP implementation in the branch](https://github.com/openthread/openthread/tree/feature/ncp-spinel/src/ncp) has been tested, and is working well enough for basic connectivity, but is far from complete. A Host implementation is not yet provided in the tree, but will likely eventually be included. A separate host implementation for Linux has been written and is anticipated to be released as a separate project over the next few weeks.

The [`feature/ncp-spinel`](https://github.com/openthread/openthread/tree/feature/ncp-spinel) branch will be under heavy development and rebases should not be unexpected.

Thoughts and feedback are welcome and appreciated.

Searaphim, 2016-05-18:
From my understanding, this is meant to be a link between whatever drives your RF radio and Thread right? Do you think this could work well with linux-wpan? http://wpan.cakelab.org/

darconeous, 2016-05-18:
Actually, Spinel is a protocol for the link between your host controller and a Thread stack that lives on a combined microcontroller+radio (Commonly called a System-on-a-Chip, or *SoC*)<sup>1</sup>, like an EM357. You wouldn't use the Spinel protocol to talk directly to an 802.15.4 radio.

If your Linux box is directly connected to an 802.15.4 radio instead of an SoC, then you'd need to run OpenThread on Linux and write up a new PHY layer to talk to whatever radio you are using. How exactly that would all work is still TBD.

We will be open-sourcing a tool shortly that will be the host-side application for communicating with an SoC/NCP that speaks the Spinel protocol.

I wasn't previously aware of [linux-wpan](http://wpan.cakelab.org/), thanks for mentioning it. It looks like it is doing the 6LoWPAN stuff on the host-side, but an SoC (Acting as an NCP) typically completely isolates the host controller from the 6LoWPAN details, providing actual real IPv6 packets directly. This allows the host controller to go to sleep while the NCP can continue to route mesh traffic.

-----

1. This is actually just one of many possible use cases, but I didn't want to confuse matters by describing all possible use cases.

Searaphim, 2016-05-19:
Thanks for the clarification. My radio is indeed directly connected to my linux box via SPI. But since linux-wpan already communicates with my radio and does the PHY, MAC, 6LowPAN layers I'm looking for ways to merge that with OpenThread. I was thinking maybe cutting up layers on either OpenThread  or linux-wpan could be the solution but lacking the documentation (including thread specifications) makes it hard to decide.

darconeous, 2016-05-19:
I could imagine `linux-wpan` being updated to use OpenThread for implementing Thread support. That would likely be the most logical path forward for getting a linux device with a directly-connected 802.15.4 radio (as opposed to a directly connected NCP) on a Thread network.

luwannest, 2016-05-20:
RQ, is there any tool like wpantund/wpanctl running with NCP app only on laptop, but not porting to EM35x or any other hardware boards? Then we could do some testings without the hardware, like what we can do with SOC app.

closed 2016-05-24 by darconeous

tammyleino, 2016-08-24:
Can someone kindly point me to any documentation on the Spinel component?  The original link is no longer active.  Thanks in advance.

turon, 2016-08-25:
Hi Tammyleino,

The protocol documentation is here:
https://github.com/openthread/openthread/blob/master/src/ncp/PROTOCOL.md

Is that what you're looking for?

Martin

_____________________________
Martin Turon  |  Nest Labs