openthread/openthread #34: Thread Sniffer and Host Side Controller

luwannest, 2016-05-15:
We will need a Thread Sniffer for debugging and analyzing the OpenThread.
The PRs or Thread Sniffer will come soon.

darconeous, 2016-05-17:
This functionality could be added to the NCP implementation, tracked by #31. Writing a simple sniffer host app that talks to the NCP should be fairly straightforward.

luwannest, 2016-05-17:
The sniffer will not implement any serial protocol. The plan is to make it being able to run on any protocols which can be NCP or any others.

darconeous, 2016-05-18:
I assume that to mean the sniffer would use the OpenThread API directly. That being the case, you still have to get the sniffed data off of the device, using some sort of protocol.

jwhui, 2016-05-19:
I agree it will be useful to include sniffer capability directly in the NCP app.

luwannest, 2016-05-19:
So will there be an option to enable NCP app or Sniffer app?
Is it clearer to have a standalone sniffer app?
The sniffer only needs Hdlc/Flen frame protocol, but does not need Spinel, and the interface to OT stack.

turon, 2016-05-19:
Hi Lu,

I think both are useful.  Certainly a dedicated sniffer firmware that can
stream into wireshark with a minimal Python tool is useful.

Being able to tap into a sniffer channel via a stock NCP app is useful as
well.

Martin

darconeous, 2016-05-19:
You need more than just HDLC or FLEN for a dedicated sniffer app. You need some way to set the channel, etc. If this capability is going to be built into the NCP (which I plan to do), might as well use that.

For example,  you can turn any spinel-based NCP into a sniffer (not considering the framing protocol) by sending it the following commands:

* `CMD_RESET`
    * Bytes: `80 01`
* `CMD_SET_PROP(PROP_PHY_ENABLED, 1)`
    * Bytes: `80 03 20 01`
* `CMD_SET_PROP(PROP_PHY_RAW_STREAM_ENABLED, 1)`
    * Bytes: `80 03 27 01`
* `CMD_SET_PROP(PROP_PHY_PROMISCUOUS_MODE_ENABLED, 1)`
    * Bytes: `80 03 28 01`
* `CMD_SET_PROP(PROP_PHY_CHAN, <Channel-Number>)`
    * Bytes: `80 03 21 XX` (where `XX` is the channel number in hex)

Then you would get raw frames on the given channel via updates to `PROP_STREAM_RAW`. The frames from the NCP would look like this:

    80 06 71 LL LL RR ... RR MM ... MM

Where:

* `LL LL` is the length of the raw packet (`XX ... XX`) in bytes, stored as a little-endian uint16.
* `RR ... RR` is the bytes of the raw packet.
* `MM ... MM` is optional metadata about the raw packet (Channel, RSSI, LQI, etc). Format TBD.

So, all things considered, it's really not any more complicated to talk to the NCP than it would be to talk to a dedicated sniffer app. Initialization is just sending the NCP a fixed set of frames, setting the last byte of the last frame to the channel. Then you handle any frame you receive starting with `80 06 71` as a received packet, and extract the packet bytes and metadata.

Note that this isn't actually implemented in the NCP yet, but doing so wouldn't be very difficult and is absolutely on my list of capabilities I intend to add.

luwannest, 2016-05-19:
RQ, thanks for your clarifications!
Could I know what is your plan for wpantund?
The host scripts would have 2 options:
- capture the packets from wpan interface working with wpantund
- implement part of Spinel and capture the packet from serial port directly

darconeous, 2016-05-19:
I was imagining a lightweight script or tool that would talk to the NCP exclusively.

Adding sniffing to wpantund could certainly be done at some point, but is currently not a feature that is planned. If there is demand for it that could change.

turon, 2016-05-19:
Yes, the sniffer is the first example of many showing we'll want the
host-side of Spinel to be easy to integrate into focused scripts such as
this sniffer python adapter for streaming into a file or into wireshark.
Making sure the code is structured such that both the client and host side
can be built as dedicated library archives will help here.  I believe
that's your plan, RQ, yes?

Martin

luwannest, 2016-05-20:
RQ, the host tool talk to the NCP that can be the sniffer host tool or any others is in your plan and you will finish it by your own. Yes?

darconeous, 2016-05-24:
@martin wrote:

> Yes, the sniffer is the first example of many showing we'll want the host-side of Spinel to be easy to integrate into focused scripts such as this sniffer python adapter for streaming into a file or into wireshark.

Indeed.

> Making sure the code is structured such that both the client and host side can be built as dedicated library archives will help here.  I believe that's your plan, RQ, yes?

There is a set of functions that I have written in `spinel.c` that are useful for parsing the Spinel data stream, but they are by no means required. The most complicated part of an implementation is likely the HDLC-lite framing.

@luwannest wrote:

> RQ, the host tool talk to the NCP that can be the sniffer host tool or any others is in your plan and you will finish it by your own. Yes?

No. At least, not at the moment. I hadn't considered or planned on writing the sniffer host app myself.

darconeous, 2016-05-26:
I will probably end up writing a small dedicated sniffer host app which talks to the NCP as soon as the API's are in place to allow sniffing.

darconeous, 2016-09-28:
By the way, basic packet capture capability was added to OpenThread and wpantund. I'd still like to have a dedicated sniffer app, so I'm keeping this open for that.

turon, 2016-09-28:
@darconeous Do you mean dedicated sniffer firmware, host utility, or both?

jwhui, 2016-09-28:
@turon, see the previous comment from @darconeous.

turon, 2016-09-28:
@jwhui Ah yes "app" in isolation is vague as to whether it is host-side or firmware.  It sounds like the proposed solution is a dedicated host-side utility that talks to stock ncp firmware.  This is something I could take on if others don't have time. @darconeous, the initialization instructions in your May 19th comment look sufficient to get started.

darconeous, 2016-09-28:
Host utility.

PR #773 (2016-10-14): ncp: Added sniffer host tool.
Split common code from spinel-cli.py out into a general spinel python library module.
Added sniffer.py tool that leverages the new spinel library module.
Added some initial unit tests of spinel python library module using MockStream.

From openthread root:

```
    sudo ./tools/spinel-cli/sniffer.py -c 11 -n 1 -u /dev/ttyUSB0 | wireshark -k -i -
```

This will connect to stock openthread ncp firmware over the given UART, make the node into a promiscuous mode sniffer on the given channel, open up wireshark, and start streaming packets into wireshark.  No preference [... cut]
closed 2016-10-14 by jwhui