openthread/openthread #37: Cut initial release

beriberikix, 2016-05-16:
When the time is right, we should cut an initial [release](https://help.github.com/articles/creating-releases/) (maybe after I/O?) Items I think we could include in the release description can include:
* Supported Thread spec version
* Spec features added, enhanced or changed
* Device, library or framework support
* Docs and tools improvements

Anything else?

darconeous, 2016-05-18:
In case the plans were to do something different, I recommend we release as a 0.1.0 release (rather than, for example, a 1.0.0 release).

We should also make sure that `make distcheck` works. At the moment travis is just doing a `make check`. Will file an issue.

darconeous, 2016-05-18:
I've gone ahead and created a 0.1.0 milestone. We can rename it if the consensus is for the initial release is to be a different version number.

jwhui, 2016-05-19:
I have no strong opinions on the milestone.  The current .default-version file has 0.5, but I'm open to change.  Any other opinions?

mszczodrak, 2016-05-19:
0.5 is already released. let the contributions grow before the number
starts changing

jwhui, 2016-05-19:
@mszczodrak , we have not officially cut a release yet.  The initial commit to GitHub is separate from actually cutting a release.

darconeous, 2016-05-19:
Given that we have not actually made a release under `0.5`, I recommend we change `.default-version` to `0.1.0` before we release, so that we can use that as our version starting point. Nice and clean.

jwhui, 2016-06-03:
@darconeous, want to submit a PR for changing the `.default-version`?

darconeous, 2016-06-11:
Sure, I'll go ahead and do that.

By the way, do you think we should move to two-digits for the minor and patch versions? The advantage of doing that is that it preserves file sorting order when either moves above 10.

jwhui, 2016-06-11:
Yes, two digits is a good idea.

darconeous, 2016-06-23:
We should write up what is the criteria for our initial release. "Gut feel" isn't going to cut it, because that is not a goal to work toward. We should have a checklist of capabilities/features that we want for 0.01.00 and set some internal soft deadlines.

jwhui, 2016-06-23:
One concrete goal could be passing all of the network-layer tests in the Thread 1.0 Certification Harness.  I'll see if there's a feasible timeline for that.

Gr3yh0und, 2017-05-04:
Are there any plans to introduce a versioning / naming scheme in the future?

jwhui, 2017-05-04:
@Gr3yh0und, we're getting very close to cutting the first release that is used as a reference by the [Thread Test Harness](http://graniteriverlabs.com/thread/).  As stated in the [README.md](https://github.com/openthread/openthread/blob/master/README.md), the plan is to follow the [Semantic Versioning Guidelines](http://semver.org/).

beriberikix, 2017-12-26:
Closed with this [release](https://github.com/openthread/openthread/releases/tag/thread-reference-20170716)?

jwhui, 2018-01-03:
@beriberikix, sounds good to me!

closed 2018-01-03 by beriberikix