openthread/openthread #44: MAC possibly allowing non-encrypted packets to be passed off as legit network packets

darconeous, 2016-05-18:
Maybe I'm misinterpreting the code, but having a look at `ProcessReceiveSecurity()` in `src/core/mac/mac.cpp`:

```
ThreadError Mac::ProcessReceiveSecurity(const Address &aSrcAddr, Neighbor *aNeighbor)
{
    ThreadError error = kThreadError_None;

    // ...

    if (mReceiveFrame.GetSecurityEnabled() == false)
    {
        ExitNow();
    }

    // ...

exit:
    return error;
}
```

I checked later on and I don't see any obvious indication that insecure packets are given any additional attention. Obviously the stack needs to be able to handle insecure traffic, but it must handle it with care to ensure that these sorts of packets aren't confused as legit network traffic and passed up the stack (except under very specific circumstances, and even then always handled differently).

jwhui, 2016-05-21:
I agree the security checks are lacking.  I plan to put a whitelist check in place where unsecure frames MUST:

1. Must be a 6LoWPAN frame that matches either 6LoWPAN IPHC or Fragment Header types.
2. IPv6 message length is limited to maximum required for MLE and Thread Commissioning.
3. IPv6 source and destination addresses MUST have link-local scope.
4. IPv6 Next Header MUST be UDP.
5. UDP Destination Port MUST match MLE UDP port and Joiner Port for Thread Commissioning.

It would be nice to also limit the total allocation of receive message buffers for unsecure frames, but it is difficult to know what a good limit is without some empirical data.

Thoughts?

jwhui, 2016-05-26:
Fixed in #68.

closed 2016-05-26 by jwhui