openthread/openthread #47: All certification tests fail on OS X

darconeous, 2016-05-19:
It would appear that currently all certification tests fail on OS X with errors similar to the following:

```
======================================================================
FAIL: test (__main__.Cert_5_1_01_RouterAttach)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "./thread-cert/Cert_5_1_01_RouterAttach.py", line 66, in test
    self.assertEqual(self.nodes[ROUTER].get_state(), 'router')
AssertionError: 'leader' != 'router'

----------------------------------------------------------------------
Ran 1 test in 10.105s

```

jwhui, 2016-05-19:
I just tried 'make check' on OS X on the current HEAD (91cce9) and the tests pass for me.  Can you check if you have any existing 'soc' processes running?  I know that interrupting the python check scripts sometimes does not properly clean up  the 'soc' processes.

jwhui, 2016-05-19:
I just submitted #48 that adds osx and it seems to be passing on travis.

darconeous, 2016-05-19:
I bet that was what was messing it up (other SoC processes running). Will try again tomorrow.

darconeous, 2016-05-19:
Still having trouble running it from my machine (might be a zombie ncp process, will try again after a reboot), but if you got it running in Travis via #48, then I'm sure it is a configuration issue on my end.

Closing.

closed 2016-05-19 by darconeous