openthread/openthread #56: OpenThread should never create or join a partition with a PANID of 0xFFFF

darconeous, 2016-05-20:
The current default value of the PANID (as retrieved by `otGetPanId()`) is `0xFFFF`. This itself isn't a bad default—it effectively implies that the PANID is undefined (because it is not a valid PANID for a network).

However, the current OpenThread stack seems to take this value and run with it anyway, happily creating a new partition with the PANID `0xFFFF` instead of doing something more reasonable (like calculating a new PANID).

I propose that the behavior for having a PANID of `0xFFFF` should be to:

* If a suitable already partition exists and has a legal PANID, use the PANID from that partition and join it. (maybe we would want to do this anyway)
* If no suitable partition is found nearby, allocate a new PANID and use that to start a new partition.

Thoughts?

jwhui, 2016-05-21:
> The current default value of the PANID (as retrieved by otGetPanId()) is 0xFFFF. This itself isn't a bad default—it effectively implies that the PANID is undefined (because it is not a valid PANID [...]

I agree that OpenThread should not use `0xFFFF` as a valid PANID.

> I propose that the behavior for having a PANID of 0xFFFF should be to:
> 
> If a suitable already partition exists and has a legal PANID, use the PANID from that partition and join it. (maybe we would want to do this anyway)

I agree that if a device is not configured with a valid PANID and a valid partition exists, it should use that partition's PANID and attach.

> If no suitable partition is found nearby, allocate a new PANID and use that to start a new partition.

I am OK with allocating a new PANID, but I don't think that OpenThread should do this autonomously.  In particular, a new PANID should only be created when initially forming the network or when changing the PANID.  I think both scenarios should require explicit action from outside OpenThread.

darconeous, 2016-07-07:
Looking at the output from the description in [openthread/wpantund/19](https://github.com/openthread/wpantund/issues/19), I believe this is still a problem.

I'm going to bump this to a P1 because I think we should address this before the [0.01.00 milestone](https://github.com/openthread/openthread/milestone/1).

PR #405 (2016-08-17): Do not allow enabling Thread protocols when PAN ID is set to broadcast.
This change simply disallows:

1. Starting Thread protocol operation via `otThreadStart()` when the PAN ID is set to broadcast.
2. Setting the PAN ID to broadcast via `otSetPanId()` when MLE is not in the disabled state.

Resolves #56.
closed 2016-08-17 by jwhui