openthread/openthread #57: cli: CTRL-C shouldn't exit on embedded platforms

turon, 2016-05-21:
This line of code is dangerous when compiled for an embedded platform:
https://github.com/openthread/openthread/blob/master/src/cli/cli_serial.cpp#L106

Accidentally pressing CTRL-C in your CLI terminal will halt execution and require a reset.  
Perhaps it would be okay if a proper watchdog is implemented on the platform.

jwhui, 2016-05-21:
I assume @darconeous added this because he wanted an easy way to exit the emulated code.  However, I'm not sure this is necessary since `termios.c_lflag` in `serial.c` does not disable the `ISIG` flag.

@darconeous , what do you think about removing the explicit handling of `Ctrl+C` in `cli_serial.cpp`?

darconeous, 2016-05-21:
I believe I still ran into some problems. It definitely warrants further investigation to figure out a way to make this work without catching the CTRL-C code.

jwhui, 2016-05-21:
FWIW, I just tried removing the referenced CTRL-C case from `master` `HEAD` and I was able to use CTRL-C to exit the process on both macos and ubuntu.

darconeous, 2016-05-23:
Fixed by #64.

closed 2016-05-23 by darconeous