openthread/openthread #59: ncp: Redesign `src/ncp.hpp` to be more memory efficient

darconeous, 2016-05-21:
In `src/ncp.hpp`, we are currently allocating two 1500 byte buffers: one for sending, and one for receiving. This is quite wasteful. There are also several cases where outbound frames might be dropped due to the fact that the device is in the process of sending a previous frame.

abtink, 2016-06-29:
Looking a bit into this:

- On the inbound direction (rx msg over serial to NCP) we may need to keep a large enough buffer to store the HDLC output. We process the received data from the callbacks so, it's pretty straightforward.
- On the outbound direction (tx msg to serial from NCP), I think we can treat the buffer as a circular buffer and allow it to contain more than one msg at a time. We may still drop a msg if there is no room left in the buffer to send. 

Also a suggestion about class names and the way they inherit from one another:
-  We have `NcpBase` which implements spinel parsing of the messages.
- And we have `Ncp` which inherits from `NcpBase` and is mainly handling the buffering. 

I think switching the roles may be better, basically making the current `Ncp` the base class, and then have a spinel sub-class (which would implement the current `NcpBase` functionality). With this, we should be able to avoid abstract virtual methods like `OutboundFrameBegin`, ... in the base class.

Any thoughts?

darconeous, 2016-06-30:
> On the inbound direction (rx msg over serial to NCP) we may need to keep a large enough buffer to store the HDLC output. We immediately process the received data from the callbacks so, it's pretty straightforward.

The big optimization here I think might be to decode into chunks so that the received data can go right into a buffer instead of staying in an intermediate buffer.

> On the outbound direction (tx msg to serial from NCP), I think we can treat the buffer as a circular buffer and allow it to contain more than one msg at a time. We may still drop a msg if there is no room left in the buffer to send.

Circular buffer sounds reasonable, but I think the bar for dropping packets should be higher and the buffer size small---like 256 bytes or so. This limits the size of control traffic somewhat, but that should be OK, we can always increase it later. For IPv6 packets, we have the buffer object which we can keep around until we top off the send buffer for the last time.

> Also a suggestion about class names and the way they inherit from one another:

Yeah, I just inherited the naming from the previous protobufs-based NCP code. Suggestions are warranted. :)

> I think switching the roles may be better, basically making the current Ncp the base class, and then have a spinel sub-class (which would implement the current NcpBase functionality). With this, we should be able to avoid abstract virtual methods like OutboundFrameBegin, ... in the base class.

I personally think the subclassing is rather unfortunate in this case, and would advocate abandoning that approach rather than swapping the meanings. This is a big rewrite, which is why I've been hesitant to dive in.

The nice thing about the current class structure is that it is pretty straightforward to create a new subclass for supporting a new transport, like SPI.

jwhui, 2016-08-10:
Resolved with #263, #264, and #329.

closed 2016-08-10 by jwhui