openthread/openthread #60: Encounter multiple definitions error, while building the app that needs to use otPlatRadio interface, but not OT stack

luwannest, 2016-05-22:
The sniffer that does not need the OT stack, but need the otPlatRadio interface, common scheduler, message queue, etc.
Then the sniffer needs to include libopenthread.a, which causes the multiple definitions error. The mac and sniffer both use otPlatRadioSignalReceiveDone.
Do we need to have a standalone library for the common functions that does not related with the thread stack? Then other app could do not always need to include the whole ot stack, while they do not need to use it, or only part of it.

turon, 2016-05-22:
Hi Lu,

It may be good to move all such otPlat callbacks into a delegate model
where any given otPlatModule has an otPlatModuleRegister to pass a
structure of function pointers.

Alternatively, the build system can be adjusted to create a libotplatform.a
that has weak stub implementations of the callbacks, which then gets linked
correctly for any given app / platform pair.

Martin

jwhui, 2016-05-22:
There are two things to consider here:

1. The sniffer as a dedicated standalone app.
2. The core OpenThread library also providing IEEE 802.15.4 packet capture capability.

For 1, I don't think it makes any sense to have a dedicated sniffer app that links against libopenthread.a.  Instead, the dedicated sniffer app should just link against the C++ code in src/core directly.

For 2, it will be necessary to have both the full Thread stack along with sniffer capabilities compiled together with a single radio interface.

luwannest, 2016-05-22:
@jwhui , thanks for your clarification. I think that I can close this issue.

closed 2016-05-22 by luwannest