openthread/openthread #101: Inconsistent example code and documentation

GiedriusM, 2016-06-02:
POSIX example platform radio [implementation](https://github.com/openthread/openthread/blob/master/examples/posix/platform/radio.cpp) can return kThreadError_Busy in state change operations (_otPlatRadioEnable_, _otPlatRadioSleep_, _otPlatRadioIdle_, _otPlatRadioReceive_ and _otPlatRadioTransmit_), however the [documentation](https://github.com/openthread/openthread/blob/master/include/platform/radio.h#L171) does not allow such return.

darconeous, 2016-06-02:
From my experience with the code base so far, `kThreadError_Busy` seems to be the go-to generic error, unfortunately. Should definately be cleaned up.

jwhui, 2016-06-14:
Fixed in #138.

closed 2016-06-14 by jwhui