openthread/openthread #106: The example /examples/posix/app/cli/soc  CAN NOT Start .

ZetaChow, 2016-06-03:
Build success,but run /examples/posix/app/cli/soc 1 , nothing happen
OS is OS X
I tried sudo , and no effect yet

mszczodrak, 2016-06-03:
Hi,
After starting the soc app, one should be able to enter into the shell 'start', 'state', 'ipaddr'. Does any of those commands return any output in your case?

ZetaChow, 2016-06-03:
@mszczodrak Thank you! I enter 'start' and output 'done'; 
I understand it correctly, before I wait for a prompt, in fact, it worked well

closed 2016-06-03 by jwhui

darconeous, 2016-06-07:
An actual command prompt would really be helpful, as I ended up having the same reaction the first time I tried it. Something even as simple as a `>` would do the trick.

jwhui, 2016-06-07:
Good idea.  Pull request coming soon.