openthread/openthread #119: api: expose ability to capture received IEEE 802.15.4 frames

jwhui, 2016-06-08:
Add ability to capture received IEEE 802.15.4 frames for sniffing and debugging.

Capturing raw IEEE 802.15.4 frames should be possible when:
1) the Thread interface is enabled, capturing all IEEE 802.15.4 frames that are processed by OpenThread.
2) the Thread interface is disabled and capturing all IEEE 802.15.4 frames with IEEE 802.15.4 promiscuous mode enabled.

turon, 2016-06-08:
Hopefully this new API could also consider how to handle multiple MAC interfaces multiplexed into a single radio interface:  Thread secured, Alternate 15.4 secured, unsecured, ...

jwhui, 2016-06-08:
OpenThread can only be responsible for frames that are passed to it.  In other words, the received frame must be provided through `otPlatRadioReceiveDone` for it to be passed through the OpenThread API.

jwhui, 2016-06-11:
Added in #120.

closed 2016-06-11 by jwhui