openthread/openthread #121: spinel: external headers should not be included by core spinel protocol

turon, 2016-06-09:
Embedded platforms may not have access to this header or the in6_addr structure included by spinel.h:

[spinel.h]
```
 #include <arpa/inet.h>
// ...
typedef struct in6_addr spinel_ipv6addr_t;
```

turon, 2016-06-09:
Here is a somewhat related issue where ncp_base.cpp assumes being built by autotools, thus making the NCP code not portable across toolchains:

[ncp_base.cpp]
```
#include <openthread-config.h>
// ...
// line 908
        PACKAGE_NAME "/" PACKAGE_VERSION "; " __DATE__ " " __TIME__
```

darconeous, 2016-06-09:
Good catch. Will fix.

darconeous, 2016-06-09:
Will likely try to avoid using OpenThread types, so that the file can be used in other projects easily without depending on the rest of OpenThread.

jwhui, 2016-06-11:
Fixed in #133.

closed 2016-06-11 by jwhui