openthread/openthread #129: test: "fuzzing" OpenThread to provide robustness and negative stress testing

turon, 2016-06-10:
We have good positive path test coverage of OpenThread now using `make check` and the CI test suite.

Perhaps the negative path test coverage of OpenThread could be built using "fuzzing" techniques, and some of the tools available in that space.

The GDS [blog](http://blog.gdssecurity.com/labs/2015/9/21/fuzzing-the-mbed-tls-library.html) discusses the technique, good open source tools to implement it, and even have examples applying the method to an instrumented mbedtls:

> "Fuzzing is a testing technique that feeds a program under inspection with random inputs and observes its behavior. The goal of fuzzing is to discover bugs in the inspected program."

darconeous, 2016-06-15:
We should also fuzz spinel.

darconeous, 2017-06-26:
OpenThread really needs a comprehensive fuzz/fault-insertion testing plan, and this seems like the most appropriate issue to track that with.

I'm bumping this up in priority to increase visibility. Also adding the "**help wanted**" tag, since @mturon doesn't necessarily have to be the person to implement it.

Tagging @jwhui, @abtink, and @nibanks for comment.

jwhui, 2017-07-14:
I'm starting work on adding initial fuzz testing support.

PR #2014 (2017-07-20): [fuzz] add support to build fuzzing targets
This enables adding OpenThread to [OSS-Fuzz](https://github.com/google/oss-fuzz).

Resolves #129.
closed 2017-07-20 by jwhui