openthread/openthread #137: Strategy for common compilers

beriberikix, 2016-06-13:
While we're targeting an OSS toolchain, folks use a bunch of different compilers @ work. IAR seems to be a common request and one that doesn't work out of the box. See [this SO](http://stackoverflow.com/questions/37767178/openthread-compiling-errors-relating-to-prefixtlv) question as an example.

It's not feasible to target every toolchain, but we should at least be able to come up with a strategy to help people use their preferred (or required) toolchains.

turon, 2016-06-13:
This "toolchain.h" pull request that was merged this weekend is part of the solution:
    https://github.com/openthread/openthread/pull/107

 It provides toolchain / compiler abstraction macros which should solve the issue of cross-compilation of OpenThread for the following popular compilers: GCC, clang, Keil, IAR, MS Visual Studio, and SDCC.

It does not provide project files for these toolchains however, leaving autotools as the only supported method to build the library.  For developers that are able to build those project files themselves, OpenThread master should now address the problem flagged in the SO question you referenced.

jwhui, 2016-06-13:
It would be great to demonstrate support for other compilers (e.g. IAR).  @turon, are you planning to do that?

turon, 2016-06-15:
Sure, once cc2538 drivers are pushed, I can look into adding an IAR project for it.

turon, 2016-07-01:
@jwhui, #190 adds verified IAR support to the OpenThread cc2538 example.

jwhui, 2016-12-14:
We currently use the gcc `-pedantic` option to "issue all the warnings demanded by strict ISO C and ISO C++".  More info [here](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html).  We also have clang, gcc, and Visual Studio build support and checks.

closed 2016-12-14 by jwhui