openthread/openthread #144: Adding clearer documentation of hardware abstraction

Saphieron, 2016-06-14:
Starting to work with this implementation it is quite complicated to find out, how the abstraction layer is realised and used. 

Is it possible to add a more extensive explanation either in a separate file, or as inline comments for the example implementations, that the line becomes clearer where the example posix implementation ends and the common open thread core code starts?

itavero, 2016-06-14:
Some kind of test suite would also be nice.
I've ordered some CC2530 and CC2560 development boards with the plan to turn them into an NCP.
But how would one verify the implementation?
Just a simple "they connect" won't do, I guess.

jwhui, 2016-06-14:
@Saphieron, having more documentation on the platform abstraction is definitely a fair comment.  We'll try to improve those.  We'd also welcome any help from the community in improving documentation based on their implementation experience.  In general, any call to `otPlat*` APIs represents a transition between platform-specific code and the OpenThread core library.  Code in `examples/` is typically platform-specific and code in `src/core/` is the OpenThread core library.

jwhui, 2016-06-14:
@itavero, unfortunately we do not have something like a "conformance" test suite of the `otPlat*` APIs.  Examples of OpenThread running on actual IEEE 802.15.4 chips is coming soon, so hopefully that will help provide a better understanding of how the APIs work on actual hardware.  We currently use the Thread Group Certification Harness for system-level testing.

jwhui, 2017-04-28:
Published a [Porting Guide](https://github.com/openthread/openthread/wiki/Porting-Guide) on the wiki.

closed 2017-04-28 by jwhui