openthread/openthread #147: Consider removing otPlatRadioIdle() and associated logic.

pvanhorn, 2016-06-14:
Currently OT uses otPlatRadioIdle() to transition the radio between receive and transmit.  It might be more natural and straightforward if OT were to instead remove the option for a radio Idle State and instead treat the radio as operating in one of the following States {Disabled, Sleep, Receive, Transmit}.  To transmit a packet OT would then make a call to otPlatRadioTransmit() without a pre-requisite call to otPlatRadioIdle().  This also creates an assumption that if the radio is not Disabled, Transmitting or Sleeping then it must be Receiving.  This should reduce opportunities for missing received packets and simplify the OT code.

GiedriusM, 2016-06-15:
I would disagree. A lot (if not all) radios have idle state, so it makes sense that software reflects. Also hardware configuration should be done while in IDLE state, otherwise unexpected behavior might occur. The way I see the states:

- Disabled - radio is shut down, almost none current consumption (leakage current), all register/configuration information is lost.
- Sleep - radio is in low power on, minimal power consumption, most hardware is disabled (clocks, PLL's, PA's, etc.), registers/configuration is kept. Requires wake-up/warm-up routine.
- Idle - radio is configured (or being configured) and running, [almost] ready to TX/RX, small power consumption.
- TX/RX - active state, high power consumption.

pvanhorn, 2016-07-22:
I don't disagree with your statements.  The problem I have is how OT currently uses otPlatRadioIdle().  OT behaves by treating receive as a one-shot operation, meaning that OT assumes the driver will automatically transition back to Idle after delivering a single received packet.  It would be preferable if receive were treated as more persistent.  The radio would remain in receive and continue to deliver packets until OT explicitly transitioned it to a new State.  Furthermore, as it exists today OT always transitions to idle by calling otPlatRadioIdle before calling otPlatRadioTransmit.  It would be preferable if OT simply called otPlatRadioTransmit without the intermediate call to Idle.  Lastly, after transmit completes, OT should assume that the radio will transition back to receive rather than Idle.  Currently, OT makes an explicit call to otPlatRadioReceive after transmit  done.

mszczodrak, 2016-07-25:
Pull request to remove Idle at #275

jwhui, 2016-07-25:
Addressed in #275.

closed 2016-07-25 by jwhui