openthread/openthread #179: ncp: some suggestions on PROTOCOL.md

librasungirl, 2016-06-19:
The encoding symbols, 'i', 'ii' or 'U' etc,  are used in Section 5, how about bringing Section 7.1 to front or adding one sentence in Section 5 illustrating that the meaning of the encoding symbols would be described in section 7.1?
It might be a little confusing when first met with the symbols in section 5 if without some explanations or any reference introduction in advance.

BTW, a minor typo in Section 4.6, the fifth line 'followed by the value to be inserted', 'inserted' should be 'removed'

darconeous, 2016-06-20:
Thanks for the suggestions! Will review and reorganize to see how that reads (I suspect you are right).

jwhui, 2016-06-23:
Fixed in #198.

closed 2016-06-23 by jwhui