openthread/openthread #191: ncp: sys/types.h not available in some embedded toolchains

turon, 2016-06-21:
The IAR toolchain libc doesn't have sys/types.h included by spinel, and thus doesn't compile at this time.  It would be good if the spinel client code didn't have this dependency so it can easily be retargeted for a wide variety of MCUs and toolchains.

darconeous, 2016-06-21:
I can't remember what I'm using `sys/types.h` for. I'm curious what breaks if you comment it out.

turon, 2016-06-21:
Hi RQ, yes, deleting that line from spinel.h seems to build fine.

turon, 2016-06-21:
Correction, these symbols seem to be missing without it:
EOVERFLOW
EINVAL
ssize_t

jwhui, 2016-06-22:
Fixed in #193.

darconeous, 2016-06-23:
`EOVERFLOW` and `EINVAL` should come from `errno.h`, included in `spinel.c`. Maybe IAR doesn't define these errors?

~~`ssize_t` should get pulled in by `stdio.h`, which needs it for the return value of `printf()`.~~ Er, nevermind, `printf()` returns `int`. >_<

I'll just get rid of the use of `ssize_t`. Will figure out what to do about `EOVERFLOW`...

darconeous, 2016-06-23:
Reopening, based on @turon's comments.

jwhui, 2016-06-23:
Fixed in #203.

closed 2016-06-23 by jwhui