openthread/openthread #200: Tests failing on Raspberry Pi 2 B

cm17g13, 2016-06-23:
Im attempting to run Openthread on my Raspberry Pi 2 B, unfortunalty i couldn't use the builds that already existied. Using "export FORCE_UNSAFE_CONFIGURE=1 && sudo ./build" within the "openthread/third_party/nlbuild-autotools/repo/tools/packages/" directory to rebuild. When trying the CLI Example, running 2 terminals to make 2 seperate nodes, both have no knowlage of the other existing (each belives it is the leader) and can't communicate (During the ping test, the packets are sent to the correct ip but there is no reply). 

Running the "make check" all the static tests work fine, however all of the other tests fail (see below)
[stdout.txt](https://github.com/openthread/openthread/files/329622/stdout.txt)
This was the result after a couple of minutes piping the standard out of the make check to a file.

cm17g13, 2016-06-23:
Both nodes are reciving regular packets, as well as "scan" returning that they have detected each other, but sill both believe they are leader.

cm17g13, 2016-06-23:
Problem fixed, however this could be a potential bug in the code.
within "openthread/examples/posix/platform/radio.c" line 258-266 

`static inline void getExtAddress(const uint8_t *frame, otExtAddress *address) 
{
    size_t i;

    for (i = 0; i < sizeof(address); i++) {
        address->m8[i] = frame[IEEE802154_DSTADDR_OFFSET + (7 - i)];
    }
}`

as "address" is a pointer doing "sizeof" will return the number of Bytes used in the pointer, not the size of the address itself. When dealing with 64 bit pointers and addresses it workes out fine as both are the same, however, the Pi uses 32 bit pointers, so it was calculating the address length to be 32 bits, meaning that half of the address length is missing during processing.

the code should read  
`static inline void getExtAddress(const uint8_t *frame, otExtAddress *address)
{
    size_t i;

    for (i = 0; i < sizeof(s_extended_address); i++)
    {
        address->m8[i] = frame[IEEE802154_DSTADDR_OFFSET + (sizeof(s_extended_address) - 1 - i)];
    }
}`

jwhui, 2016-06-23:
Good catch!  I just submitted #201 to fix this issue.

jwhui, 2016-06-23:
Fixed in #201.

closed 2016-06-23 by jwhui