openthread/openthread #204: Adding/Removing Unicast IPv6 addresses to Netif.

abtink, 2016-06-23:
I think the current implementation of APIs for adding/removing address can cause some issues.
The method `otAddUnicastAddress(otNetifAddress *aAddress)` will actually insert the passed-in address instance (`aAddress`) at the head of the `Netif::mUnicastAddresses` linked list (without copying it).  

In the usage from `NcpBase::InsertPropertyHandler_IPV6_ADDRESS_TABLE()` the passed in instance is a local (allocated on stack) variable which will go away when we return from this function. Similarly, in `cli.cpp` from `Interpreter::ProcessIpAddrAdd` we use a common static instance `static otNetifAddress sAddress` which may be changed/overwritten.

For `otRemoveUnicastAddress(otNetifAddress *aAddress)`, we do a pointer comparision of the passed-in `aAddress` with the elements in the `mUnicastAddresses` list. I think we should check the content and IPv6 address.

Some suggestions on this:

1) We can use a fixed array to store addresses in netif and copy the added new addresses.
2) We can have a pool of addresses which are used to copy and store the addresses (forming a linked-list)
3) We can a hybrid approach, some fixed addresses which are always there like 'LinkLocal' or 'MeshLocal` address and then an array or pool for the user-added addresses.

Briefly thinking about this I lean more towards the pool of address model (even though it may be more complicated). The reason is that I think it can help us enhance how we handle the `StateChangedCallback` for event 'OT_IP6_ADDRESS_REMOVED'. It'd be good to somehow notify the callback which address(es) were removed. With the pool model we can keep the removed addresses in another linked-list which the callback then can get and clear/free afterwards.

Thoughts?

turon, 2016-06-23:
+1 for address pool.

jwhui, 2016-06-23:
The intent of the API was to have the client provide the storage needed for storing netif addresses.  The advantage of having the client provide the memory is that OpenThread itself does not have to be configured to support a certain number of addresses.  The `cli.cpp` is closest to that, the only issue is that `cli.cpp` does not check to make sure that the static struct was already added.

The address pool/link-list approach is least appealing to me.  If we go with OpenThread-allocated netif structures, then we might as well have a fixed array and save the cost of storing/maintaining pointers.

darconeous, 2016-06-23:
The nice thing that we got from the table+linked-list was that it allowed changes to be more easily communicated, but admittedly this is not a critical capability. Leaving it up to the client to allocate the structures is valuable. However, I seem to keep flip-flopping on which one is more important.

In any case, if we keep this behavior we should make sure it is pretty clear what is happening in the documentation.

Due to my misunderstanding of the behavior of the API, the existing code for adding/removing addresses is rather broke. This could be fixed by adding a table of addresses to the `NcpBase` class.

jwhui, 2016-06-30:
Fixed in #219.

closed 2016-06-30 by jwhui