openthread/openthread #213: cli: any plan to add children/routes/routers clis?

librasungirl, 2016-06-27:
are we going to add children/routes/routers clis to display current children set, routing table and router set?  might be useful for observing and evaluating the network status.

jwhui, 2016-06-29:
Yup, I've been meaning to add these.  Watch for a PR soon.

librasungirl, 2016-06-30:
Thanks @jwhui .  Looking forward to your PR.  Some more supplements here just as remind:
1) cache - show current EID-to-RLOC map 
2) whitelist - show current whitelist configuration

librasungirl, 2016-07-01:
Thanks @jwhui  for these PRs. Supplement one more cli:
3) leaderdata - show current partition information (leaderid, partitionid, network data version, network data stable version etc)

jwhui, 2016-07-04:
Resolved by #225, #226, $228, #229, and #231.

closed 2016-07-04 by jwhui