openthread/openthread #216: Possible issue in `Leader::ExternalRouteLookup()`

abtink, 2016-06-27:
I am not sure if this is really an issue/bug, but thought it'd be safer to mention it so it can be checked... :)

In `Leader::ExternalRouteLookup()` method, we go through the network data tlvs in `mTlvs` and check any `PrefixTlv` to see if prefix matches our destination, and if it does we then go through sub-tlvs of the prefix tlv and check for preference, rout cost , etc.

We use the variable `cur` in an outer loop to to go through all the tlvs and then in an inner for loop we use `cur` to go through the prefix sub-tlvs.  If I am not mistaken, at the end of the inner sub-tlv loop, the `cur` will be at the end of all sub-tlvs of the `PrefixTlv` which should be start of next tlv. When/If we go back to the outer loop, we will do a `cur = cur->GetNext()` as part of the loop iteration which is intended to jump over the `PrefixTlv` but I think it will skip the next tlv. @jwhui  is there something I am missing? Thanks.

If this is actually an issue, fix should be simple: We can use a different local var like `subCur` for inner loop.

jwhui, 2016-06-27:
Yes, good catch!  Feel free to submit a PR to fix this.  Thanks.

jwhui, 2016-06-28:
Fixed in #217.

closed 2016-06-28 by jwhui