openthread/openthread #230: ncp: openthread-config.h.in assumes autotools and isn't toolchain portable

turon, 2016-07-01:
The iar port of the openthread apps isn't able to find openthread-config.h since it is assumed to be autogenerated by autotools.  This will be a roadblock for building NCP firmware on other toolchains as well.

jwhui, 2016-07-01:
The only items used out of `openthread-config.h` is `PACKAGE_NAME` and `PACKAGE_VERSION`.  Those can easily be generated as necessary:
* `PACKAGE_NAME` is just `OPENTHREAD`
* `PACKAGE_VERSION` is generated with the bash script `third_party/nlbuild-autotools/repo/scripts/mkversion`

closed 2016-07-01 by jwhui

turon, 2016-07-01:
Wouldn't it be simpler to just have a static .h for those then?  
It seems best to minimize the friction for folks trying to compile NCP with other toolchains.

jwhui, 2016-07-01:
A static header file would not capture the actual state of your git tree, which is the purpose of the `mkversion` bash script.