openthread/openthread #238: Mesh routing problem

GiedriusM, 2016-07-04:
Imagine the following setup: 3 devices A, B and C. A is in reach of B, but out of reach of C and vice-versa. Create network (leader) on node B, then join A and C to B as routers. Using mesh local addresses ping can pass A->B, B->A, B->C, C->B, but not A->C or C->A. The first showstopper I found is when A wants to send a packet to unknown mesh-local adress, it sends address query to B and AddressResolver::HandleAddressQuery is executed, however no response is sent because neither the mesh-C address matches mesh-B (obvious), nor any child of B exists with mesh-C address. To be more precise B->children list is empty. The reason for this is that A and C are removed from children list when A/C->MleRouter::HandleAddressSolicitResponse is called, their short address is chaged from "child-short" to "router-short", LinkRequest sent to B and B->MleRouter::HandleLinkRequest is executed (A and C are removed from B->children when handling Tlv::kSourceAddress).
Without having access to Thread specification I am unsure how this scenario should be handled. Should A and C be left in the B->children list or AddressQuery go past routers or maybe just pass through the AddressQuerry to other routers?
P.S. It seems that multicast addresses are also not retransmitted to other routers/nodes.

GiedriusM, 2016-07-04:
I forgot to mention, that I use NCP when testing.
As for the multicast issue, it looks like the retransmission is implemented only in MeshForwarder::HandleMesh, but not in MeshForwarder::HandleLowpanHC and ICMP (and other) frames that originated from host (and not Thread device) are processed in the latter function. Am I right, or missing something? How should host frames be handled - retransmitted as Lowpan frames or converted to Mesh frames before the first transmission?

turon, 2016-07-04:
Hi Giedrius,

Yes, given OpenThread requires you to input shared network parameters to
start a network.  These pre-shared parameters minimally include the network
key and the xpanid.  I call this minimal set of parameters the Out-Of-Band
or OOB parameters.  Yes, it tends to be easier to setup a network when the
Mesh Local Prefix is derivable somehow, which has the added benefit of
keeping the minimum set of OOB parameters small. There are three ways to
remedy your problem:

1) include the Mesh Local Prefix in the OOB parameters used to setup nodes
A,B, and C.
2) derive the Mesh Local Prefix from some known parameter such as the
XPANID delivered in the Beacon.
3) get the Mesh Local Prefix during the attaching process.

Martin

_____________________________
Martin Turon  |  Nest Labs

GiedriusM, 2016-07-04:
Hi, Martin,
I don't think I follow what you are saying. As far as I understand, you are saying that the nodes need to have a mesh local address, however the nodes in my setup have mesh local addresses and they can ping each other as long as they are within direct radio signal reach. The problem occurs when two of them (A and C) are too far away and need to route packets via the midpoint device (B). When A wants to send a frame to an unknown mesh address (mesh-C) it issues an address query to B, however B does not respond, because it has no child entry with such address.

turon, 2016-07-04:
Hi Giedrius,

Address Query is a concept that is only valid within a single Thread
Network where all nodes by definition must share the same Mesh Local Prefix.

Are you saying you are trying to use Address Query on a Thread Network
where node A and C have different Mesh Local Prefixes?

If you want to use other prefixes in Thread, you need to add a Border
Router and use GUA addresses.  The is always exactly one Mesh Local Prefix
which must be shared by all nodes on the mesh.

Martin

_____________________________
Martin Turon  |  Nest Labs

jwhui, 2016-07-04:
@turon, you are prematurely jumping to conclusions and confusing the issue.  Let me try to address the issues directly.

> Imagine the following setup: 3 devices A, B and C. A is in reach of B, but out of reach of C and vice-versa. Create network (leader) on node B, then join A and C to B as routers. Using mesh local [...]

Do all three of your nodes have the same Mesh Local prefix?  If so, the Address Query should propagate from A to C.  Also, when a device becomes a router

> To be more precise B->children list is empty. The reason for this is that A and C are removed from children list when A/C->MleRouter::HandleAddressSolicitResponse is called, their short address is [...]
Without having access to Thread specification I am unsure how this scenario should be handled. Should A and C be left in the B->children list or AddressQuery go past routers or maybe just pass through the AddressQuerry to other routers?

This is correct.  When a device becomes a router, it is no longer a child of any other router.

> P.S. It seems that multicast addresses are also not retransmitted to other routers/nodes.

The Address Query should propagate across multiple routers.

jwhui, 2016-07-04:
@GiedriusM, I just tested your scenario with the `ot-cli` example app and it seems to work for me.  This is what I did:

Node B: 
```bash
$ ot-cli 2
> extaddr
e2b3540590b0fd87
Done
> whitelist add f1d92a82c8d8fe43
Done
> whitelist add d38d7f875888fccb
Done
> whitelist enable
Done
> start
Done
> state
leader
Done
```

Node C:
```bash
$ ot-cli 3
> extaddr
d38d7f875888fccb
Done
> whitelist add e2b3540590b0fd87
Done
> whitelist enable
Done
> start
Done
> state
router
Done
> state
router
Done
> ipaddr
fdde:ad00:beef:0:0:ff:fe00:c400
fe80:0:0:0:0:ff:fe00:c400
fdde:ad00:beef:0:110a:e041:8399:17cd
fe80:0:0:0:d18d:7f87:5888:fccb
Done
```

Node A:
```bash
$ ot-cli 1
> extaddr
f1d92a82c8d8fe43
Done
> whitelist add e2b3540590b0fd87
Done
> whitelist enable
Done
> start
Done
> state
router
Done
> ping fdde:ad00:beef:0:0:ff:fe00:c400
16 bytes from fdde:ad00:beef:0:0:ff:fe00:c400: icmp_seq=0 hlim=64 time=47ms
```

GiedriusM, 2016-07-04:
Yes, all nodes are on the same Thread network, i.e. have same pan name, id, mesh-local prefix, etc.
It's now clear to me that the address querry should be propagated and I have a feeling that this may be the problem. I will test this tomorrow to see if this is the case and, if so, why.

jwhui, 2016-07-04:
I believe this may be a bug specific to NCP mode in OpenThread.  In particular, when a packet is forwarded to the host, it stops getting forward across the mesh.  This is not a problem for unicast packets.  However, multicast packets need to be forwarded both to the host and across the mesh.  I think this should be another simple fix.  Thanks for raising all of these issues!

GiedriusM, 2016-07-05:
I've found the problem to be in `Ip6::HandleDatagram` when a packet (address querry) has both receive and forward flags selected and the code exists right after handling receive:
```
        if (sReceiveIp6DatagramCallback != NULL && fromLocalHost == false)
        {
            sReceiveIp6DatagramCallback(&message);
            ExitNow(error = kThreadError_None);
        }
```

I've tried replacing `ExitNow(error = kThreadError_None);` with `error = kThreadError_None;` (note - error must be cleared otherwise premature Message::Free will be called), however this change leads to system hangup soon (infinite loop in `Message::Free`)  after receiving and forwarding the packet. It looks like this happens because `Message::Free` is called twice for the same message - once after handling receive and second after forwarding. There a few possible solutions here:
1) If receive and forward are both set, clone the message.
2) Add refence counter for message and only allow freeing it only if the count is zero.
3) Remove message freeing from the "receiver" and/or "forwarder" layers.

I don't think 3 is a viable option as it would require most changes, and would likely bring nasty memory management bugs. First one seems to be the easiest, however I'm a bit concerned about memory usage (how well would it work with large (1000+ byte) frames?).

jwhui, 2016-07-05:
Just submitted #243 which implements option 1 above.  It is the easiest and also happens to address #241.  But, as you say, may not be the most memory efficient.  We can optimize later if needed.

PR #243 (2016-07-07): Forward multicast datagrams when operating in NCP mode
Fixes #238 and #241.
closed 2016-07-07 by jwhui