openthread/openthread #241: Incorrect IPv6 and UDP header length fields in decompressed packets

GiedriusM, 2016-07-04:
When lowpan frames are decompressed, the constructed IPv6 and UDP header length fields include the appended CCM tag (length fields are larger by `sizeof(tag)`). In practice this is not a problem, because these raw values are not used, instead `message.GetLength()` is called to determine the payload size and it is updated correctly in `Mle::HandleUdpReceive` after the CCM decryption `SuccessOrExit(aMessage.SetLength(aMessage.GetLength() - sizeof(messageTag)));`. However this was quite confusing when I first dumped inbound packets on the host side and the buffer length did not match the header length field:

```
wpantund[24123]: [NCP->] CMD_PROP_VALUE_IS(PROP_STREAM_NET_INSECURE)           
wpantund[24123]: [NCP->] IPv6 len:87 type:17(cksum 0xd74d) [INSECURE] [DROPPED]
60 00 00 00 00 33 11 ff    fe 80 00 00 00 00 00 00                             
7e f4 85 67 ae f4 56 55    ff 02 00 00 00 00 00 00                             
00 00 00 00 00 00 00 01    4d 4c 4d 4c 00 33 d7 4d                             
00 15 a0 01 00 00 00 00    00 00 01 04 00 02 90 00                             
0b 08 38 3e d6 fd 00 02    7e 24 09 0b cd 00 00 00                             
40 08 00 00 00 f1 01  
```

This is probably not so important, as long as these frames are processed internally only.

jwhui, 2016-07-04:
@GiedriusM, MLE packets do not have MAC enabled, but instead have security enabled at the MLE layer.  Since MLE is part of the UDP payload, the tag should be included in the IPv6 and UDP length values.  What's happening here is that the MLE implementation is stripping off the tag after security processing and then the modified packet is being delivered to the host after MLE processing is done.

In general, processed MLE packets should probably not be exported to the host.  `wpantund` is just dropping them anyway, as you noted above.

GiedriusM, 2016-07-04:
Thanks! I had a similar feeling, however without having Thread specifications it's hard to determine whether something is a real bug or an attribute.

jwhui, 2016-07-05:
See #243.

jwhui, 2016-07-07:
Fixed in #243.

closed 2016-07-07 by jwhui