openthread/openthread #245: Platform abstraction for slave SPI interface

darconeous, 2016-07-06:
We need a platform abstraction for a slave SPI interface. I present the following API for consideration and discussion:

```
/**
 * Indicates that a SPI transaction has completed with
 * the given length. The data written to the slave has been
 * written to the pointer indicated by the `input_ptr` argument
 * to the previous call to `otPlatSPISlavePrepareTransaction()`.
 *
 * If this function returns `true`, the transaction is cleared
 * and a subsequent call to `otPlatSPISlavePrepareTransaction()` is
 * required for the next transaction to proceed.
 *
 * If this function returns `false`, the previously prepared
 * transaction is kept intact.
 *
 * @param[in] len     Length of the completed transaction, in bytes.
 *
 * @retval `true`  The previously prepared transaction is to be
 *                 cleared and a subsequent call to
 *                 `otPlatSPISlavePrepareTransaction()` is required
 *                 for the next transaction to proceed.
 * @retval `false` The previously prepared transaction is kept
 *                 intact.
 */
bool otPlatSPISlaveTransactionComplete(uint16_t len);

/**
 * Initialize the SPI slave interface. Not fully ready until
 * `otPlatSPISlavePrepareTransaction()` is called. If
 * `otPlatSPISlavePrepareTransaction()` is not called before
 * the master begins a transaction, the resulting SPI transaction
 * should send all `0xFF` bytes.
 *
 * @retval ::kThreadError_None  Successfully enabled the SPI Slave interface.
 * @retval ::kThreadError_Fail  Failed to enable the SPI Slave interface.
 */
ThreadError otPlatSPISlaveEnable(void);

/**
 * Shutdown and disable the SPI slave interface.
 */
void otPlatSPISlaveDisable(void);

/**
 * Prepare data for the next transaction. Data pointers
 * MUST remain valid until `otPlatSPISlaveTransactionComplete()`
 * is called by the SPI slave driver.
 * 
 * @param[in] output_ptr  Data to be written to MISO pin
 * @param[in] input_ptr   Data to be read from MOSI pin
 * @param[in] max_len     Maximum transaction length, in bytes
 *
 * @retval ::kThreadError_None  Transaction was successfully prepared.
 * @retval ::kThreadError_Busy  A transaction is currently in progress.
 */
ThreadError otPlatSPISlavePrepareTransaction(
    const uint8_t *output_ptr,
    uint8_t *input_ptr,
    uint16_t max_len
);

/**
 * Returns true is a SPI transaction is in progress.
 * Returns false otherwise.
 */
bool otPlatSPISlaveTransactionInProgress(void);

/**
 * Signals to the SPI master that we want it to
 * initiate a SPI transaction. There is no error
 * condition. The request must be asserted for
 * after the call to otPlatSPISlavePrepareTransaction(). 
 */
void otPlatSPISlaveRequestTransaction(void);
```

darconeous, 2016-07-06:
Tagging @luwannest and @turon.

darconeous, 2016-07-06:
Also tagging, @abtink to give his thoughts about the proposed API.

darconeous, 2016-07-06:
Note that this API is unsuitable as a general-purpose SPI slave API—and I really doubt that such an API would be practical anyway. But this API does work very well for the [recommended SPI protocol for Spinel](https://github.com/openthread/openthread/blob/master/src/ncp/PROTOCOL.md#a21-spi-framing-protocol), allowing it to be efficiently implemented. It would also likely work well for other SPI framing protocols.

turon, 2016-07-06:
Hi @darconeous,

Thanks for kicking off this discussion.  Top-level, there seems to be two ways to go here:

1) A dedicated otPlatSpiSlave API as you propose.  Split the inheritance tree of NcpBase as follows:

```
       NcpBase
              |
       NcpProtocolSpinel  (currently collapsed into NcpBase)
       /             \
 NcpUart    NcpSpiSlave
```

2) Keep the existing otPlatSerial API, and have the platform figure out how to handle packetization.  Keep the current NcpBase inheritence tree:
```
       NcpBase
              |
       NcpProtocolSpinel  (currently collapsed into NcpBase)
              |
       Ncp (uses generic otPlatSerial for either SPI or UART as decided by platform)
```

I had always assumed we'd go down path (2), but either are valid.  The way I had thought otPlatSerial could work with a SPI Slave implementation is as follows:

```
ThreadError otPlatSerialEnable(void) 
{
    // Enable SPI Slave bus driver, and gpios: 
    // chip select (CS), slave-to-host signal irq (TO_HOST), host-to-slave signal (FROM_HOST).
}

ThreadError otPlatSerialDisable(void) 
{
    // Opposite of Enable, as appropriate for platform.
}

ThreadError otPlatSerialSend(const uint8_t *aBuf, uint16_t aBufLength) 
{
    // Copy outbound packet to internal driver tx_buffer, signal TO_HOST gpio, and wait for master to pull data
}

void platformSpiSlaveIsr(void)
{
    // SPI transaction / CS IRQ: handle master transaction.  CS assumed to surround a complete packet.
    // Read any data from master into an internal rx_buffer.  Write data from tx_buffer out.
}

extern void otPlatSerialSendDone(void) 
{
    // Send at end of SPI transaction whenever tx_buffer is successfully sent out.
}

extern void otPlatSerialReceived(const uint8_t *aBuf, uint16_t aBufLength)
{
    // Sent up to application from task context when rx_buffer is filled.  rx_buffer may be cleared by driver when this returns.
}
```

The main question with method (2) is how is the length of tx_buffer communicated from slave to master?  A simple way to do that would be to define a short SPI header that prepends each transaction payload that would minimally send the length of the packet.  

For basic flow control, one could assume GPIOs provide direction for a half-duplex utilization of the bus  where a master essentially performs padded "read" transactions anytime TO_HOST is signaled (setting FROM_HOST to low), and performs pure "write" transactions when FROM_HOST is asserted.

Host-to-Slave
```
FROM_HOST = 1
SPI     CS:    1 0 --------------------------------- 1
SPI  HOST:     [len][data ---- ]
SPI SLAVE:    [0][0 ----- 0]
```

Slave-to-Host
```
TO_HOST = (pulse 1 to signal to master to pull data)
FROM_HOST = 0
SPI     CS:    1 0 --------------------------------- 1
SPI SLAVE:     [len][data ---- ]
SPI  HOST:    [0][0 ----- 0]
```

darconeous, 2016-07-06:
It is my professional opinion that path two is a bad idea.

SPI behaves fundamentally differently than a UART. Trying to make SPI work like a UART is wasteful on many levels. It wastes memory, it wastes performance, and it wastes engineering time.

The SPI subclass to `NcpBase` (or `NcpProtocolSpinel` as you call it) will be more simple than the current implementation of `Ncp` (or `NcpUart`, as it will likely be renamed).

turon, 2016-07-06:
It would be good to isolate specifically where a path (1) API benefits memory, performance, and engineering time if only to document the justification of choice for the decision.

My take on the answers here are:
  Memory / Performance: NcpSpiSlave will be in charge of all data buffers and the flow-control header, so can compose those into single DMA friendly frames.
  Engineering Time: platform/spi_slave.c driver focuses on a one-to-one mapping of what a typical hw datasheet does.  NcpSpiSlave handles the special flow-control header so properly separates that out from the lower-layer driver.

Some minor review comments on your proposed API:
- nit: I believe our the OpenThread convention is to camel case abbreviations, so otPlatSpiSlave rather than otPlatSPISlave.  This can be seen in MleRouter, NcpBase, and other examples.
- the calls that are from the driver up to OpenThread should be marked extern until we move to function pointer callbacks.
- otPlatSPISlaveRequestTransaction can be a parameter of otPlatSPISlavePrepareTransaction.

Thanks for the proposal!

darconeous, 2016-07-06:
Will apply your bulleted comments.

darconeous, 2016-07-06:
Have a look at #248 for the pull request.

jwhui, 2016-07-07:
Addressed by #248.

closed 2016-07-07 by jwhui