openthread/openthread #252: CC2538:  no ack for data request from SED

librasungirl, 2016-07-07:
One device as Leader, one as SED,  Leader would not ack the data request from SED

platform CC2538DK
firmware openthread master branch (commitid ef41589)

[image]

jwhui, 2016-07-07:
@librasungirl, can you see if this was fixed by #247?

librasungirl, 2016-07-07:
Thanks, will try out

librasungirl, 2016-07-07:
hi @jwhui, here is the update for the try with the latest master branch(ab28c3a) - include #247
 
The first Data Request could be acked, but afterwards, SED would hang and trapped in below code,

`// in func ThreadError otPlatRadioTransmit(void) of cc2538 radio driver
while ((HWREG(RFCORE_XREG_FSMSTAT1) & 1) == 0);`

the traffic is as below
[image]

Will update if any new process.

GiedriusM, 2016-07-08:
Offtopic:
@librasungirl Do you use any custom wireshark dissector for MLE (or maybe even for Thread?). If not, could you provide your wireshark version information. I'm having trouble getting MLE packets parsed in my pcaps.

librasungirl, 2016-07-09:
hi @GiedriusM , the special wireshark comes along with GRL's Test Harness. You may need to download  and install Harness first from http://graniteriverlabs.com/thread/ and then configure wireshark according to the ‘Quick Start Guide' before you use it to dissect thread packets. 
Please have a try.

librasungirl, 2016-07-09:
The root cause is explained as below, FYI.
After handled Child Id Response in upper layer, before go back to radio driver, NextOperation() in mac layer would call otPlatformSleep() and the typo ` VerifyOrExit(error = kStateIdle, error = kThreadError_Busy);` would force the status of the radio be kStateSleep, so when go down to radio driver,  disableReceiver() would not take effect; RX FIFO is not flushed, that is why  `(HWREG(RFCORE_XREG_FSMSTAT1)` is observed as 0x54 (overflow) when SED device is trying to send another packet but trapped in `while ((HWREG(RFCORE_XREG_FSMSTAT1) & 1) == 0);` 

PR#256 has beed created for this.

jwhui, 2016-07-09:
Fixed in #256.

closed 2016-07-09 by jwhui