openthread/openthread #266: regarding building openthread in native pc

Praveen-Kumar-K, 2016-07-13:
Hi all,
I am new to thread, i wanted to test CLI application on my linux PC, for that i have followed README in CLI folder, where after "make -f examples/Makefile-posix" command, it is saying build for node1, node2, for which a posix folder should be generated in above command i am unable to view this folder. Can any one help me in this regard and is it possible to test in native PC if so please guide me.

Thanks,
Praveen Kumar k

jwhui, 2016-07-13:
@Praveen-Kumar-K, the README file is out-of-date.  Thanks for the notice - I'll submit an update to the README.  After a successful build, you should see a new `openthread/output` directory and the executables in `openthread/output/<platform>/bin`

librasungirl, 2016-07-14:
hi @jwhui , I think it might not only about the doc.
Under my environment(4.2.0-38-generic #45~14.04.1-Ubuntu), I have to use sudo before 'make -f examples/Makefile-cc2538' or 'make -f examples/Makefile-posix' or else there would be 'Permission denied' issue; and the binaries would be in the absolute path '/output/bin/arm-none-eabi-ot-cli' and '/output/x86_64-unknown-linux-gnu/bin/ot-cli' respectively.

` make -f examples/Makefile-cc2538 
  BUILD    cortex-m3-arm-none-eabi
make -C build/cortex-m3-arm-none-eabi --no-print-directory all
  CHECK    .local-version
/bin/sh: .local-version.N: Permission denied
make[1]: *** [.local-version] Error 1
make: *** [do-build-cortex-m3] Error 2

make -f examples/Makefile-posix
  BUILD    x86_64-unknown-linux-gnu
make -C build/x86_64-unknown-linux-gnu --no-print-directory all
  CHECK    .local-version
/bin/sh: .local-version.N: Permission denied
make[1]: *** [.local-version] Error 1
make: *** [do-build-x86_64-unknown-linux-gnu] Error 2
`

PR #267 (2016-07-15): docs: update path to executables
Fixes #266.
closed 2016-07-15 by jwhui