openthread/openthread #271: Creating Thread network in simulator

Praveen-Kumar-K, 2016-07-20:
Hi all,
I want to simulate Thread network on my PC, i have followed wpantund tutorial mentioned in repo.
I have used 4 clients (1 Leader and 3 routers assigned), one Wpantund, one utun6.
clients haveing parameters as shown:
cl1:  name----wpantund-test; ExPANID-----deba7ab1e5eaf00a; PAN-----FFFF; Chaneel---11;
cl2:  name----wpantund-test; ExPANID-----deba7ab1e5eaf00b; PAN-----FFFF; Chaneel---11; -leader
cl3:  name----wpantund-test; ExPANID-----deba7ab1e5eaf00c; PAN-----FFFF; Chaneel---11;
cl4:  name----wpantund-test; ExPANID-----deba7ab1e5eaf00d; PAN-----FFFF; Chaneel---11;

For Wpantund terminal i have done command.
-------------------------------------------------------------------------
sudo /usr/local/sbin/wpantund -o Config:NCP:SocketPath "system:examples/apps/ncp/ot-ncp 2" -o SyslogMask " -info" -o Config:TUN:InterfaceName utun6
[sudo] password for pkamma: 
wpantund[320]: Starting wpantund 0.07.00 (Jul 13 2016 17:54:13) . . .
wpantund[320]: Reading configuration from "/etc/wpantund.conf" . . .
wpantund[320]: Configuration file "/etc/wpantund.conf" read.
wpantund[320]: CONFIGURATION PROPERTY "SyslogMask" IS DEPRECATED. Please use "Daemon:SyslogMask" instead.
wpantund[320]: set-config-param: "Config:NCP:SocketPath" = "system:examples/apps/ncp/ot-ncp 2"
wpantund[320]: set-config-param: "Config:TUN:InterfaceName" = "utun6"
wpantund[320]: set-config-param: "Daemon:SyslogMask" = " -info"
wpantund[320]: Ready. Using DBUS bus ":1.21"
wpantund[322]: About to exec "examples/apps/ncp/ot-ncp 2"
wpantund[320]: Running as root without dropping privileges!
wpantund[320]: State change: "uninitialized" -> "offline"
wpantund[320]: NCP is running "OPENTHREAD/0.01.00; Jul 14 2016 10:59:06"
wpantund[320]: Driver is running "0.07.00 (; Jul 13 2016 17:54:13)"
wpantund[320]: "fe80::e0b3:5405:90b0:fd87" was added to "utun6"
wpantund[320]: Network Name: OpenThread
**wpantund[320]: **Error fetching property 96 from NCP: 2****
wpantund[320]: Finished initializing NCP
-------------------------------------------------------------------------------

in Utun6 terminal: When applying scan i am not able to see clients.

To proceed further, to form a network what should i do, can any one help me in this regard.

Thanks,
Pravee Kumar K

darconeous, 2016-07-22:
Are those devices supposed to be on the same network? If so, you need to make sure they all have the same XPANID. As in, they should all be `deba7ab1e5eaf00d`, for example.

Praveen-Kumar-K, 2016-07-29:
Hello darconeous,

By keeping all with same XPANID, I am not able to form as network. BY command SCAN in terminal it is not displaying any network credentials. can any one tell me the reason and what would my mistake.

Thanks & Regards,
Praveen Kumar K

darconeous, 2016-08-02:
As you can probably tell, development on this codebase is moving forward at a furious pace and are pretty swamped at the moment.

We will try to get to verifying the tutorial within the next few days.

Sorry for the delay!

Praveen-Kumar-K, 2016-08-03:
Thanks @darconeous, mean while i will also try to verify and resolve it.
Let me know how leader can send a message to client (control Client from leader/ router), could you tell me the procedure for establishing communication between two nodes and exchange data between them.
Regards,
Praveen Kumar K

shahab5, 2016-08-22:
Hi,

Did you guys found a way to fix this?
I get the same error too.

wpanctl is used to scan, WPANTUND reports a segmentation fault.

**** WPANTUND *****
```
wpantund[12043]: Starting wpantund 0.07.00 (Aug 22 2016 10:24:32) . . .
wpantund[12043]: 	SOURCE_VERSION = 0.07.00-3-gc0ffa2a-dirty
wpantund[12043]: 	BUILD_VERSION = 0.07.00-3-gc0ffa2a-dirty
wpantund[12043]: Reading configuration from "/etc/wpantund.conf" . . .
wpantund[12043]: Configuration file "/etc/wpantund.conf" read.
wpantund[12043]: CONFIGURATION PROPERTY "SyslogMask" IS DEPRECATED. Please use "Daemon:SyslogMask" instead.
wpantund[12043]: set-config-param: "Config:NCP:SocketPath" = "system:examples/apps/ncp/ot-ncp 2"
wpantund[12043]: set-config-param: "Config:TUN:InterfaceName" = "utun6"
wpantund[12043]: set-config-param: "Daemon:SyslogMask" = " -info"
wpantund[12043]: Ready. Using DBUS bus ":1.88"
wpantund[12044]: About to exec "examples/apps/ncp/ot-ncp 2"
wpantund[12043]: Running as root without dropping privileges!
wpantund[12043]: [-NCP-]: NCP was reset (112)
wpantund[12043]: [-NCP-]: NCP is using different protocol minor version (NCP:1, wpantund:0)
wpantund[12043]: State change: "uninitialized" -> "offline"
wpantund[12043]: NCP is running "OPENTHREAD/g7629974; Aug 22 2016 10:18:34"
wpantund[12043]: Driver is running "0.07.00 (0.07.00-3-gc0ffa2a-dirty; Aug 22 2016 10:24:32)"
wpantund[12043]: "fe80::e0b3:5405:90b0:fd87" was added to "utun6"
wpantund[12043]: Network Name: OpenThread
wpantund[12043]: Error fetching property 96 from NCP: 2
wpantund[12043]: Finished initializing NCP

 *** FATAL ERROR: Caught signal 11 (Segmentation fault):
/usr/local/sbin/wpantund[0x8078cb3]
linux-gate.so.1(__kernel_rt_sigreturn+0x0)[0xb7719bb0]
/usr/local/libexec/wpantund/ncp-spinel.so(_ZN2nl8wpantund17SpinelNCPTaskScan14vprocess_eventEiPc+0x6e9)[0xb72bc4d9]
/usr/local/libexec/wpantund/ncp-spinel.so(_ZN2nl8wpantund17SpinelNCPInstance14vprocess_eventEiPc+0xb6)[0xb72b1876]
/usr/local/sbin/wpantund(_ZN2nl12EventHandler13process_eventEiz+0x19)[0x80a33e9]
/usr/local/libexec/wpantund/ncp-spinel.so(_ZN2nl8wpantund17SpinelNCPInstance32handle_ncp_spinel_value_insertedE17spinel_prop_key_tPKhj+0x3a)[0xb72a470a]
/usr/local/libexec/wpantund/ncp-spinel.so(_ZN2nl8wpantund17SpinelNCPInstance26handle_ncp_spinel_callbackEjPKhj+0x1b3)[0xb72a4a23]
/usr/local/libexec/wpantund/ncp-spinel.so(_ZN2nl8wpantund17SpinelNCPInstance18ncp_to_driver_pumpEv+0x58b)[0xb72ac34b]
/usr/local/sbin/wpantund(_ZN2nl8wpantund15NCPInstanceBase7processEv+0xe1)[0x809af11]
/usr/local/sbin/wpantund(main+0xdec)[0x8077ddc]
/lib/libc.so.6(__libc_start_main+0xf5)[0xb730a715]
/usr/local/sbin/wpantund[0x8078ae6]
wpantund[12043]:  *** FATAL ERROR: Caught signal 11 (Segmentation fault):
wpantund[12043]: [BT]  0: /usr/local/sbin/wpantund() [0x8078cb3]
wpantund[12043]: [BT]  1: linux-gate.so.1(__kernel_rt_sigreturn+0) [0xb7719bb0]
wpantund[12043]: [BT]  2: /usr/local/libexec/wpantund/ncp-spinel.so(_ZN2nl8wpantund17SpinelNCPTaskScan14vprocess_eventEiPc+0x6e9) [0xb72bc4d9]
wpantund[12043]: [BT]  3: /usr/local/libexec/wpantund/ncp-spinel.so(_ZN2nl8wpantund17SpinelNCPInstance14vprocess_eventEiPc+0xb6) [0xb72b1876]
wpantund[12043]: [BT]  4: /usr/local/sbin/wpantund(_ZN2nl12EventHandler13process_eventEiz+0x19) [0x80a33e9]
wpantund[12043]: [BT]  5: /usr/local/libexec/wpantund/ncp-spinel.so(_ZN2nl8wpantund17SpinelNCPInstance32handle_ncp_spinel_value_insertedE17spinel_prop_key_tPKhj+0x3a) [0xb72a470a]
wpantund[12043]: [BT]  6: /usr/local/libexec/wpantund/ncp-spinel.so(_ZN2nl8wpantund17SpinelNCPInstance26handle_ncp_spinel_callbackEjPKhj+0x1b3) [0xb72a4a23]
wpantund[12043]: [BT]  7: /usr/local/libexec/wpantund/ncp-spinel.so(_ZN2nl8wpantund17SpinelNCPInstance18ncp_to_driver_pumpEv+0x58b) [0xb72ac34b]
wpantund[12043]: [BT]  8: /usr/local/sbin/wpantund(_ZN2nl8wpantund15NCPInstanceBase7processEv+0xe1) [0x809af11]
wpantund[12043]: [BT]  9: /usr/local/sbin/wpantund(main+0xdec) [0x8077ddc]
wpantund[12043]: [BT] 10: /lib/libc.so.6(__libc_start_main+0xf5) [0xb730a715]
wpantund[12043]: [BT] 11: /usr/local/sbin/wpantund() [0x8078ae6]
```

jwhui, 2016-09-19:
@Praveen-Kumar-K @shahab5, are you still running into this issue?

shahab5, 2016-09-20:
Hi Jonathan,

With the latest code wpantund joins the network without any issues. I can
scan and join.

However, while forming networking as the first node, the xpanid is
incorrectly used, I have reported another issue on GitHub for this.

Thanks for your concern.

Regards
Shahab

jwhui, 2016-09-20:
Thanks, closing this issue.

closed 2016-09-20 by jwhui