openthread/openthread #272: Possible issue with adding border router prefix

abtink, 2016-07-20:
While testing some of the interaction with the OT and wpantund in simulator, I noticed that I could not  get border router prefix config to work. It may be known problem, but decided to create this issue to track this. 

Here are the steps: 

1)  Setup the simulator wpantund and ot as in the wiki.
2)  Tried adding a prefix using wpanctl which seemed to go through ok. 
```
wpanctl:utun6> config-gateway fd00:beef:: 
Using prefix "fd00:beef::"
Gateway configured.
```
3) The logs on wpanctud look ok: 
```
wpantund[61518]: Inbound DBus message for INTERFACE "ConfigGateway" from ":1.58"
wpantund[61518]: [->NCP] CMD_PROP_VALUE_INSERT(SPINEL_PROP_THREAD_ON_MESH_NETS)
wpantund[61518]: 	↳ {84045AFD00BEEF00000000E0B3540590B0FD87400030}
wpantund[61518]: SELECT: fd_count=7 cms_timeout=5000
wpantund[61518]: SELECT:     read_fd_set: 3, 4, 5, 6
wpantund[61518]: SELECT:     write_fd_set: 
wpantund[61518]: SELECT: fd_count=7 cms_timeout=5000
wpantund[61518]: SELECT:     read_fd_set: 3, 4, 5, 6
wpantund[61518]: SELECT:     write_fd_set: 
wpantund[61518]: [NCP->] CMD_PROP_VALUE_INSERTED(SPINEL_PROP_THREAD_ON_MESH_NETS)
wpantund[61518]: Sending DBus response for "ConfigGateway" to ":1.58"
wpantund[61518]: SELECT: fd_count=7 cms_timeout=60000
wpantund[61518]: SELECT:     read_fd_set: 3, 4, 5, 6
wpantund[61518]: SELECT:     write_fd_set: 
```
4) After this expecting to see the new address based on the prefix but we don't:
```
wpanctl:utun6> get ipv6:alladdresses
ipv6:alladdresses = [
	"fdde:ba7a:b1e5:0:9704:6a73:208:7028  valid: 4294967295  preferred: 4294967295  flags: ---- ---- (0x00)"
	"fe80::e0b3:5405:90b0:fd87  valid: 4294967295  preferred: 4294967295  flags: ---- ---- (0x00)"
]
```
I tried doing the same using CLI with similar outcome:
```
> prefix add fd00:1234::/64
Done
> ipaddr
fdde:ba7a:b1e5:0:0:ff:fe00:3000
fe80:0:0:0:0:ff:fe00:3000
fdde:ba7a:b1e5:0:9248:3362:41f3:d23
fe80:0:0:0:f3d9:2a82:c8d8:fe43
Done
```

jwhui, 2016-07-20:
I just tried it out using the CLI and it works for me.  Two issues I noted with your CLI example:
1. Need to pass in the `v` flag to indicate that this is a valid SLAAC prefix.
2. Need to register the network data with the leader.  OT separates this step out to aggregate Network Data updates to the leader.

```
> start
Done
> prefix add fd00:1234::/64 v
Done
> netdataregister
Done
> ipaddr
fd00:1234:0:0:48e9:c5f1:b0c4:158a
fdde:ad00:beef:0:0:ff:fe00:9c00
fe80:0:0:0:0:ff:fe00:9c00
fdde:ad00:beef:0:558:f56b:d688:799
fe80:0:0:0:f3d9:2a82:c8d8:fe43
Done
```

abtink, 2016-07-20:
Thanks @jwhui . It works with cli  as you said :) 
I guess on wpantund side, we should also issue the "nestdataregister" command after prefix add command. Will fix this in PR in wpantund...

abtink, 2016-07-20:
Talking to @darconeous, here is our suggestion to address this:
- Will add a new spinel property called `THREAD_ALLOW_LOCAL_NET_DATA_CHANGE` which is `boolean`. 
- This property should be set to `true` before modifying any of local net data properties (e.g., inserting a new value to `ON_MESH_NETS` or a `LOCAL_ROUTES`, otherwise, we get an `SPINEL_STATUS_INVALID_STATE` error if we try to write to these properties).
- When this property transitions from `true` to `false`, we register all the net data changes with leader. 

This serves couple of purposes:
- It provide a way from wpantund to use  the flexibility of OT to update multiple net data properties before registering the changes with leader.
- It ensures only one entity is making the changes (this is not immediately applicable with current wpantund model but may be helpful in future).

I am working on changes to address this. 

Any thoughts/suggestions?

abtink, 2016-07-21:
Two PRs with the changes:
- [PR 273](https://github.com/openthread/openthread/pull/273) in openthread.
- [PR 25](https://github.com/openthread/wpantund/pull/25) in wpantund.

abtink, 2016-07-21:
Tested/verified the two changes together.
`wpanctl`:
```
wpanctl:utun6> config-gateway fd00:1234::
Using prefix "fd00:1234::"
Gateway configured.
wpanctl:utun6> config-gateway fd00:5678::
Using prefix "fd00:5678::"
Gateway configured.
wpanctl:utun6> get ipv6:alladdresses
ipv6:alladdresses = [
	"fd00:1234::fa6d:a0d3:6f03:1f87  valid: 4294967295  preferred: 4294967295  flags: ---- ---- (0x00)"
	"fd00:5678::6bc4:999e:ba3e:10cf  valid: 4294967295  preferred: 4294967295  flags: ---- ---- (0x00)"
	"fdab:cdef::9704:6a73:208:7028  valid: 4294967295  preferred: 4294967295  flags: ---- ---- (0x00)"
	"fe80::e0b3:5405:90b0:fd87  valid: 4294967295  preferred: 4294967295  flags: ---- ---- (0x00)"
]
```

jwhui, 2016-07-21:
Thanks, @abtink.  I agree with the approach.

abtink, 2016-07-22:
Both PRs are merged in. Let's close this issue.

closed 2016-07-22 by abtink