openthread/openthread #295: otPlatRadioSetExtendedAddress provides opposite endianness to otGetExtendedAddress

Ciaran54, 2016-07-28:
in Mac::setExtAddress, the endianness of the extAddress is flipped before being forwarded to the platform layer. However, if otGetExtendedAddress() is called, the address is provided unflipped.

This seems like an inconsistency, and I am unsure which option is big and which is little endian.

jwhui, 2016-07-28:
The APIs should be consistent.
1. OpenThread Client API: `otGetExtendedAddress` and `otSetExtendedAddress` take/provide the EUI-64 where the OUI is the first bytes.
2. OpenThread Platform API: `otPlatRadioSetExtendedAddress` takes the EUI-64 where the OUI is the last bytes.  Currently, `otPlatRadioGetExtendedAddress` does not exist.  The radios I have worked with all expect the OUI in the last bytes and is consistent with the IEEE 802.15.4 header encoding, which is why the Platform API uses this order.

Ciaran54, 2016-07-28:
Ah yes, thanks for the clarification, that does make sense, thanks

closed 2016-07-28 by Ciaran54