openthread/openthread #308: bootstrapping to build on ARM fails

martinzsmith, 2016-07-31:
```
$ ./bootstrap
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/scripts/bootstrap: line 180: /home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/armv7l-unknown-linux-gnueabihf/bin/libtoolize: No such file or directory
chmod: cannot access ‘/tmp/tmp.bootstrapvaD57j/libtoolize’: No such file or directory
Can't locate object method "good_version" via package "Autom4te::C4che" at /usr/bin/autom4te line 1002.
aclocal: error: echo failed with exit status: 25

$ ls /home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/*/bin/*
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/i686-pc-cygwin/bin/aclocal
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/i686-pc-cygwin/bin/aclocal-1.14
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/i686-pc-cygwin/bin/autoconf
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/i686-pc-cygwin/bin/autoheader
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/i686-pc-cygwin/bin/autom4te
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/i686-pc-cygwin/bin/automake
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/i686-pc-cygwin/bin/automake-1.14
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/i686-pc-cygwin/bin/autoreconf
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/i686-pc-cygwin/bin/autoscan
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/i686-pc-cygwin/bin/autoupdate
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/i686-pc-cygwin/bin/cygltdl-7.dll
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/i686-pc-cygwin/bin/ifnames
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/i686-pc-cygwin/bin/libtoolize
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/i686-pc-cygwin/bin/m4.exe
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/i686-pc-cygwin/bin/make.exe
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/i686-pc-linux-gnu/bin/aclocal
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/i686-pc-linux-gnu/bin/aclocal-1.14
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/i686-pc-linux-gnu/bin/autoconf
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/i686-pc-linux-gnu/bin/autoheader
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/i686-pc-linux-gnu/bin/autom4te
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/i686-pc-linux-gnu/bin/automake
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/i686-pc-linux-gnu/bin/automake-1.14
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/i686-pc-linux-gnu/bin/autoreconf
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/i686-pc-linux-gnu/bin/autoscan
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/i686-pc-linux-gnu/bin/autoupdate
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/i686-pc-linux-gnu/bin/ifnames
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/i686-pc-linux-gnu/bin/libtool
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/i686-pc-linux-gnu/bin/libtoolize
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/i686-pc-linux-gnu/bin/m4
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/i686-pc-linux-gnu/bin/make
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/x86_64-apple-darwin/bin/aclocal
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/x86_64-apple-darwin/bin/aclocal-1.14
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/x86_64-apple-darwin/bin/autoconf
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/x86_64-apple-darwin/bin/autoheader
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/x86_64-apple-darwin/bin/autom4te
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/x86_64-apple-darwin/bin/automake
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/x86_64-apple-darwin/bin/automake-1.14
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/x86_64-apple-darwin/bin/autoreconf
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/x86_64-apple-darwin/bin/autoscan
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/x86_64-apple-darwin/bin/autoupdate
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/x86_64-apple-darwin/bin/cp
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/x86_64-apple-darwin/bin/ifnames
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/x86_64-apple-darwin/bin/libtool
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/x86_64-apple-darwin/bin/libtoolize
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/x86_64-apple-darwin/bin/m4
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/x86_64-apple-darwin/bin/make
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/x86_64-unknown-linux-gnu/bin/aclocal
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/x86_64-unknown-linux-gnu/bin/aclocal-1.14
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/x86_64-unknown-linux-gnu/bin/autoconf
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/x86_64-unknown-linux-gnu/bin/autoheader
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/x86_64-unknown-linux-gnu/bin/autom4te
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/x86_64-unknown-linux-gnu/bin/automake
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/x86_64-unknown-linux-gnu/bin/automake-1.14
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/x86_64-unknown-linux-gnu/bin/autoreconf
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/x86_64-unknown-linux-gnu/bin/autoscan
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/x86_64-unknown-linux-gnu/bin/autoupdate
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/x86_64-unknown-linux-gnu/bin/cp
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/x86_64-unknown-linux-gnu/bin/ifnames
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/x86_64-unknown-linux-gnu/bin/libtool
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/x86_64-unknown-linux-gnu/bin/libtoolize
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/x86_64-unknown-linux-gnu/bin/m4
/home/openthread/src/openthread/third_party/nlbuild-autotools/repo/tools/host/x86_64-unknown-linux-gnu/bin/make
$ uname -m
armv7l
$
```

itsbert, 2016-07-31:
Never mind just found this:
https://github.com/openthread/openthread/blob/master/third_party/nlbuild-autotools/repo/README.md

martinzsmith, 2016-07-31:
Nevermind just found this:
https://github.com/openthread/openthread/blob/master/third_party/nlbuild-autotools/repo/README.md

closed 2016-07-31 by martinzsmith