openthread/openthread #346: Harness Outputs

librasungirl, 2016-08-08:
This is not an issue but a trick to upload harness outputs.

librasungirl, 2016-08-08:
[Leader_5_1_1.zip](https://github.com/openthread/openthread/files/406105/Leader_5_1_1.zip)
[Router_5_1_1.zip](https://github.com/openthread/openthread/files/406106/Router_5_1_1.zip)
[Router_5_1_2.zip](https://github.com/openthread/openthread/files/406107/Router_5_1_2.zip)
[Router_5_1_3.zip](https://github.com/openthread/openthread/files/406108/Router_5_1_3.zip)
[Router_5_1_4.zip](https://github.com/openthread/openthread/files/406109/Router_5_1_4.zip)
[Router_5_1_6.zip](https://github.com/openthread/openthread/files/406111/Router_5_1_6.zip)
[Router_5_1_8.zip](https://github.com/openthread/openthread/files/406110/Router_5_1_8.zip)
[Router_5_1_9.zip](https://github.com/openthread/openthread/files/406112/Router_5_1_9.zip)
[Router_5_1_10.zip](https://github.com/openthread/openthread/files/406113/Router_5_1_10.zip)
[Router_5_1_11.zip](https://github.com/openthread/openthread/files/406114/Router_5_1_11.zip)
[Router_5_1_12.zip](https://github.com/openthread/openthread/files/406115/Router_5_1_12.zip)
[Router_5_1_13.zip](https://github.com/openthread/openthread/files/406116/Router_5_1_13.zip)

[Router_5_2_1.zip](https://github.com/openthread/openthread/files/406122/Router_5_2_1.zip)
[Router_5_3_1.zip](https://github.com/openthread/openthread/files/406123/Router_5_3_1.zip)

[Router_5_3_3.zip](https://github.com/openthread/openthread/files/406125/Router_5_3_3.zip)
[Leader_5_3_8.zip](https://github.com/openthread/openthread/files/406128/Leader_5_3_8.zip)

[Router_5_5_1.zip](https://github.com/openthread/openthread/files/406130/Router_5_5_1.zip)
[Router_5_5_2.zip](https://github.com/openthread/openthread/files/406132/Router_5_5_2.zip)
[Router_5_5_3.zip](https://github.com/openthread/openthread/files/406133/Router_5_5_3.zip)
[Router_5_5_4.zip](https://github.com/openthread/openthread/files/406131/Router_5_5_4.zip)

[Leader_5_5_7.zip](https://github.com/openthread/openthread/files/406136/Leader_5_5_7.zip)
[Router_5_5_7.zip](https://github.com/openthread/openthread/files/406134/Router_5_5_7.zip)

[Router_5_6_1.zip](https://github.com/openthread/openthread/files/406137/Router_5_6_1.zip)
[Router_5_6_3.zip](https://github.com/openthread/openthread/files/406138/Router_5_6_3.zip)
[Leader_5_6_4.zip](https://github.com/openthread/openthread/files/406140/Leader_5_6_4.zip)
[Leader_5_6_5.zip](https://github.com/openthread/openthread/files/406141/Leader_5_6_5.zip)

librasungirl, 2016-08-08:
[ED_6_1_1.zip](https://github.com/openthread/openthread/files/406144/ED_6_1_1.zip)
[SED_6_1_1.zip](https://github.com/openthread/openthread/files/406150/SED_6_1_1.zip)
[ED_6_1_2.zip](https://github.com/openthread/openthread/files/406143/ED_6_1_2.zip)
[SED_6_1_2.zip](https://github.com/openthread/openthread/files/406152/SED_6_1_2.zip)
[ED_6_1_3.zip](https://github.com/openthread/openthread/files/406145/ED_6_1_3.zip)
[SED_6_1_3.zip](https://github.com/openthread/openthread/files/406153/SED_6_1_3.zip)
[ED_6_1_4.zip](https://github.com/openthread/openthread/files/406147/ED_6_1_4.zip)
[SED_6_1_4.zip](https://github.com/openthread/openthread/files/406155/SED_6_1_4.zip)
[ED_6_1_5.zip](https://github.com/openthread/openthread/files/406146/ED_6_1_5.zip)
[SED_6_1_5.zip](https://github.com/openthread/openthread/files/406158/SED_6_1_5.zip)
[ED_6_1_6.zip](https://github.com/openthread/openthread/files/406148/ED_6_1_6.zip)
[SED_6_1_6.zip](https://github.com/openthread/openthread/files/406159/SED_6_1_6.zip)
[ED_6_1_7.zip](https://github.com/openthread/openthread/files/406149/ED_6_1_7.zip)

[ED_6_2_1.zip](https://github.com/openthread/openthread/files/406161/ED_6_2_1.zip)
[SED_6_2_1.zip](https://github.com/openthread/openthread/files/406165/SED_6_2_1.zip)
[ED_6_2_2.zip](https://github.com/openthread/openthread/files/406164/ED_6_2_2.zip)
[SED_6_2_2.zip](https://github.com/openthread/openthread/files/406166/SED_6_2_2.zip)

[ED_6_3_1.zip](https://github.com/openthread/openthread/files/406167/ED_6_3_1.zip)

[ED_6_4_2.zip](https://github.com/openthread/openthread/files/406168/ED_6_4_2.zip)

[ED_6_5_1.zip](https://github.com/openthread/openthread/files/406170/ED_6_5_1.zip)

[ED_6_5_2.zip](https://github.com/openthread/openthread/files/406171/ED_6_5_2.zip)
[SED_6_5_2.zip](https://github.com/openthread/openthread/files/406175/SED_6_5_2.zip)

[ED_6_5_3.zip](https://github.com/openthread/openthread/files/406172/ED_6_5_3.zip)
[SED_6_5_3.zip](https://github.com/openthread/openthread/files/406176/SED_6_5_3.zip)

[ED_6_6_1.zip](https://github.com/openthread/openthread/files/406173/ED_6_6_1.zip)
[SED_6_6_1.zip](https://github.com/openthread/openthread/files/406177/SED_6_6_1.zip)

[ED_6_6_2.zip](https://github.com/openthread/openthread/files/406174/ED_6_6_2.zip)
[SED_6_6_2.zip](https://github.com/openthread/openthread/files/406178/SED_6_6_2.zip)

librasungirl, 2016-08-08:
[Leader_7_1_1.zip](https://github.com/openthread/openthread/files/406179/Leader_7_1_1.zip)

[Router_7_1_2.zip](https://github.com/openthread/openthread/files/406181/Router_7_1_2.zip)

[Leader_7_1_3.zip](https://github.com/openthread/openthread/files/406180/Leader_7_1_3.zip)

[Router_7_1_4.zip](https://github.com/openthread/openthread/files/406183/Router_7_1_4.zip)

librasungirl, 2016-08-22:
Harness Release 14.0 with OpenThread (commitid 7629974, master branch on GitHub)

[Leader_5_1_1.zip](https://github.com/openthread/openthread/files/429884/Leader_5_1_1.zip)
[Router_5_1_1.zip](https://github.com/openthread/openthread/files/429885/Router_5_1_1.zip)
[Router_5_1_2.zip](https://github.com/openthread/openthread/files/429889/Router_5_1_2.zip)
[Router_5_1_3.zip](https://github.com/openthread/openthread/files/429890/Router_5_1_3.zip)
[Router_5_1_4.zip](https://github.com/openthread/openthread/files/429891/Router_5_1_4.zip)
[Leader_5_1_5.zip](https://github.com/openthread/openthread/files/429892/Leader_5_1_5.zip)
[Router_5_1_6.zip](https://github.com/openthread/openthread/files/429893/Router_5_1_6.zip)
[Router_5_1_8.zip](https://github.com/openthread/openthread/files/429895/Router_5_1_8.zip)
[Router_5_1_9.zip](https://github.com/openthread/openthread/files/429894/Router_5_1_9.zip)
[Router_5_1_10.zip](https://github.com/openthread/openthread/files/429898/Router_5_1_10.zip)
[Router_5_1_11.zip](https://github.com/openthread/openthread/files/429896/Router_5_1_11.zip)
[Router_5_1_12.zip](https://github.com/openthread/openthread/files/429897/Router_5_1_12.zip)
[Router_5_1_13.zip](https://github.com/openthread/openthread/files/429899/Router_5_1_13.zip)

[Router_5_2_1.zip](https://github.com/openthread/openthread/files/429902/Router_5_2_1.zip)

[Router_5_3_1.zip](https://github.com/openthread/openthread/files/429905/Router_5_3_1.zip)
[Router_5_3_2.zip](https://github.com/openthread/openthread/files/429907/Router_5_3_2.zip)
[Router_5_3_3.zip](https://github.com/openthread/openthread/files/429906/Router_5_3_3.zip)
[Router_5_3_4.zip](https://github.com/openthread/openthread/files/429909/Router_5_3_4.zip)
[Router_5_3_5.zip](https://github.com/openthread/openthread/files/429908/Router_5_3_5.zip)
[Leader_5_3_6.zip](https://github.com/openthread/openthread/files/429910/Leader_5_3_6.zip)
[Leader_5_3_7.zip](https://github.com/openthread/openthread/files/429911/Leader_5_3_7.zip)
[Leader_5_3_8.zip](https://github.com/openthread/openthread/files/429912/Leader_5_3_8.zip)
[Router_5_3_9.zip](https://github.com/openthread/openthread/files/429915/Router_5_3_9.zip)
[Router_5_3_10.zip](https://github.com/openthread/openthread/files/429914/Router_5_3_10.zip)

[Leader_5_5_1.zip](https://github.com/openthread/openthread/files/429926/Leader_5_5_1.zip)
[Router_5_5_1.zip](https://github.com/openthread/openthread/files/429927/Router_5_5_1.zip)

[Router_5_5_2.zip](https://github.com/openthread/openthread/files/429929/Router_5_5_2.zip)
[Leader_5_5_3.zip](https://github.com/openthread/openthread/files/429931/Leader_5_5_3.zip)
[Router_5_5_3.zip](https://github.com/openthread/openthread/files/429930/Router_5_5_3.zip)

[Leader_5_5_7.zip](https://github.com/openthread/openthread/files/429933/Leader_5_5_7.zip)
[Router_5_5_7.zip](https://github.com/openthread/openthread/files/429934/Router_5_5_7.zip)

[Router_5_6_1.zip](https://github.com/openthread/openthread/files/429935/Router_5_6_1.zip)
[Leader_5_6_2.zip](https://github.com/openthread/openthread/files/429937/Leader_5_6_2.zip)
[Router_5_6_3.zip](https://github.com/openthread/openthread/files/429940/Router_5_6_3.zip)
[Leader_5_6_4.zip](https://github.com/openthread/openthread/files/429941/Leader_5_6_4.zip)
[Leader_5_6_5.zip](https://github.com/openthread/openthread/files/429942/Leader_5_6_5.zip)
[Leader_5_6_6.zip](https://github.com/openthread/openthread/files/429943/Leader_5_6_6.zip)
[Router_5_6_9.zip](https://github.com/openthread/openthread/files/429949/Router_5_6_9.zip)

[Router_5_8_2.zip](https://github.com/openthread/openthread/files/429950/Router_5_8_2.zip)

librasungirl, 2016-08-22:
[ED_6_1_1.zip](https://github.com/openthread/openthread/files/430086/ED_6_1_1.zip)
[ED_6_1_2.zip](https://github.com/openthread/openthread/files/430087/ED_6_1_2.zip)
[ED_6_1_3.zip](https://github.com/openthread/openthread/files/430089/ED_6_1_3.zip)
[ED_6_1_4.zip](https://github.com/openthread/openthread/files/430088/ED_6_1_4.zip)
[ED_6_1_5.zip](https://github.com/openthread/openthread/files/430091/ED_6_1_5.zip)
[ED_6_1_6.zip](https://github.com/openthread/openthread/files/430090/ED_6_1_6.zip)
[ED_6_1_7.zip](https://github.com/openthread/openthread/files/430092/ED_6_1_7.zip)

[ED_6_2_1.zip](https://github.com/openthread/openthread/files/430093/ED_6_2_1.zip)
[ED_6_2_2.zip](https://github.com/openthread/openthread/files/430094/ED_6_2_2.zip)

[ED_6_3_1.zip](https://github.com/openthread/openthread/files/430095/ED_6_3_1.zip)

[ED_6_4_1.zip](https://github.com/openthread/openthread/files/430096/ED_6_4_1.zip)
[ED_6_4_2.zip](https://github.com/openthread/openthread/files/430097/ED_6_4_2.zip)

[ED_6_5_1.zip](https://github.com/openthread/openthread/files/430098/ED_6_5_1.zip)
[ED_6_5_2.zip](https://github.com/openthread/openthread/files/430099/ED_6_5_2.zip)
[ED_6_5_3.zip](https://github.com/openthread/openthread/files/430100/ED_6_5_3.zip)

[ED_6_6_1.zip](https://github.com/openthread/openthread/files/430101/ED_6_6_1.zip)
[ED_6_6_2.zip](https://github.com/openthread/openthread/files/430102/ED_6_6_2.zip)

librasungirl, 2016-08-22:
[Leader_7_1_1.zip](https://github.com/openthread/openthread/files/430112/Leader_7_1_1.zip)
[Router_7_1_2.zip](https://github.com/openthread/openthread/files/430113/Router_7_1_2.zip)

[Router_7_1_5.zip](https://github.com/openthread/openthread/files/430114/Router_7_1_5.zip)

librasungirl, 2016-08-25:
Harness Release 16.0 with OpenThread (commitid 97cbae2, master branch on GitHub)
[MED_6_3_2.zip](https://github.com/openthread/openthread/files/436089/MED_6_3_2.zip)
[SED_6_3_2.zip](https://github.com/openthread/openthread/files/436090/SED_6_3_2.zip)

librasungirl, 2016-08-26:
verified PR#457 (eaf22f6) with Harness Release 14.0
[Router_7_1_8.zip](https://github.com/openthread/openthread/files/438634/Router_7_1_8.zip)

librasungirl, 2016-08-30:
Verification with Harness Release 17.0, OpenThread(04564ac)
[Leader_5_5_1.zip](https://github.com/openthread/openthread/files/444930/Leader_5_5_1.zip)
[Leader_5_6_2.zip](https://github.com/openthread/openthread/files/444931/Leader_5_6_2.zip)
[Router_5_3_5.zip](https://github.com/openthread/openthread/files/444932/Router_5_3_5.zip)
[Router_5_8_3.zip](https://github.com/openthread/openthread/files/444933/Router_5_8_3.zip)

librasungirl, 2016-08-31:
Verification with Harness Release 18.0, OpenThread(72bd784)
[Leader_5_3_7.zip](https://github.com/openthread/openthread/files/446821/Leader_5_3_7.zip)
[Router_5_3_11.zip](https://github.com/openthread/openthread/files/446822/Router_5_3_11.zip)

librasungirl, 2016-09-01:
[PR#480](https://github.com/openthread/openthread/pull/480) verification with Harness Release 18.0
[Router_7_1_7.zip](https://github.com/openthread/openthread/files/448725/Router_7_1_7.zip)

xiaom-GitHub, 2016-09-06:
Verification with Harness Release 19.0, OpenThread(1e9d3c1)
[Router_5_3_7.zip](https://github.com/openthread/openthread/files/456340/Router_5_3_7.zip)

xiaom-GitHub, 2016-09-06:
[Router_5_1_7.zip](https://github.com/openthread/openthread/files/456353/Router_5_1_7.zip)
[Leader_5_2_3.zip](https://github.com/openthread/openthread/files/456354/Leader_5_2_3.zip)
[REED_5_2_4.zip](https://github.com/openthread/openthread/files/456355/REED_5_2_4.zip)
[REED_5_2_5.zip](https://github.com/openthread/openthread/files/456356/REED_5_2_5.zip)
[REED_5_2_7.zip](https://github.com/openthread/openthread/files/456357/REED_5_2_7.zip)
[REED_5_5_5.zip](https://github.com/openthread/openthread/files/456358/REED_5_5_5.zip)
[REED_5_6_7.zip](https://github.com/openthread/openthread/files/456362/REED_5_6_7.zip)
[Router_5_2_6.zip](https://github.com/openthread/openthread/files/456364/Router_5_2_6.zip)
[Router_5_2_7.zip](https://github.com/openthread/openthread/files/456365/Router_5_2_7.zip)

xiaom-GitHub, 2016-09-11:
[REED_5_2_3.zip](https://github.com/openthread/openthread/files/466293/REED_5_2_3.zip)

xiaom-GitHub, 2016-09-14:
[REED_5_5_5.zip](https://github.com/openthread/openthread/files/471198/REED_5_5_5.zip)

luwannest, 2016-09-14:
[Leader_9_2_3.zip](https://github.com/openthread/openthread/files/471544/Leader_9_2_3.zip)

luwannest, 2016-09-20:
[Leader_9_2_1.zip](https://github.com/openthread/openthread/files/482263/Leader_9_2_1.zip)

luwannest, 2016-09-20:
[Commissioner_9_2_1.zip](https://github.com/openthread/openthread/files/482314/Commissioner_9_2_1.zip)

luwannest, 2016-09-21:
[Leader_9_2_2.zip](https://github.com/openthread/openthread/files/484184/Leader_9_2_2.zip)

luwannest, 2016-09-21:
[Commissioner_9_2_2.zip](https://github.com/openthread/openthread/files/484689/Commissioner_9_2_2.zip)

luwannest, 2016-09-23:
[Leader_9_2_4.zip](https://github.com/openthread/openthread/files/489134/Leader_9_2_4.zip)

librasungirl, 2016-09-26:
PR#685 with Harness V23
[Router_5_3_2.zip](https://github.com/openthread/openthread/files/492739/Router_5_3_2.zip)

PR#652 Harness V23
[Router_5_7_1.zip](https://github.com/openthread/openthread/files/492741/Router_5_7_1.zip)

librasungirl, 2016-09-26:
for rawgit test
[NewPdf_Router__2016_09_26-17_35_27.pdf](https://github.com/openthread/openthread/files/492833/NewPdf_Router__2016_09_26-17_35_27.pdf)

luwannest, 2016-09-27:
[Leader_9_2_5.zip](https://github.com/openthread/openthread/files/494495/Leader_9_2_5.zip)

librasungirl, 2016-09-27:
33dba76 with Harness V23
[Router_5_6_9.zip](https://github.com/openthread/openthread/files/495052/Router_5_6_9.zip)

xiaom-GitHub, 2016-09-27:
b851e92 with Harness V20
[FED_9_2_13.zip](https://github.com/openthread/openthread/files/495143/FED_9_2_13.zip)
[MED_9_2_13.zip](https://github.com/openthread/openthread/files/495144/MED_9_2_13.zip)
[Router_9_2_13.zip](https://github.com/openthread/openthread/files/495145/Router_9_2_13.zip)
[SED_9_2_13.zip](https://github.com/openthread/openthread/files/495147/SED_9_2_13.zip)

chshu, 2016-09-28:
[Joiner_Router_8_2_5.zip](https://github.com/openthread/openthread/files/498309/Joiner_Router_8_2_5.zip)

luwannest, 2016-09-29:
ef4ebfaf with Harness V1.1 R20
[Leader_9_2_1.zip](https://github.com/openthread/openthread/files/500950/Leader_9_2_1.zip)
[Leader_9_2_2.zip](https://github.com/openthread/openthread/files/500951/Leader_9_2_2.zip)
[Leader_9_2_3.zip](https://github.com/openthread/openthread/files/500953/Leader_9_2_3.zip)
[Commissioner_9_2_1.zip](https://github.com/openthread/openthread/files/500954/Commissioner_9_2_1.zip)
[Commissioner_9_2_2.zip](https://github.com/openthread/openthread/files/500956/Commissioner_9_2_2.zip)

xiaom-GitHub, 2016-09-29:
537f0fd with Harness V1.1 R20
[Commissioner_9_2_14.zip](https://github.com/openthread/openthread/files/501154/Commissioner_9_2_14.zip)
[Router_9_2_14.zip](https://github.com/openthread/openthread/files/501155/Router_9_2_14.zip)

chshu, 2016-10-03:
[Joiner_Router_8_2_1.zip](https://github.com/openthread/openthread/files/506067/Joiner_Router_8_2_1.zip)
[Joiner_Router_8_2_2.zip](https://github.com/openthread/openthread/files/506069/Joiner_Router_8_2_2.zip)

xiaom-GitHub, 2016-10-09:
3ebef25 with Harness V1.1 R26
[Router_5_1_7.zip](https://github.com/openthread/openthread/files/518141/Router_5_1_7.zip)

xiaom-GitHub, 2016-10-11:
3ebef25 with Harness V1.1 R26
[Leader_9_2_7.zip](https://github.com/openthread/openthread/files/520803/Leader_9_2_7.zip)

lluwang, 2016-10-11:
[ED_9_2_6.zip](https://github.com/openthread/openthread/files/520862/ED_9_2_6.zip)
[Leader_9_2_6.zip](https://github.com/openthread/openthread/files/520863/Leader_9_2_6.zip)

lluwang, 2016-10-11:
[SED_9_2_6.zip](https://github.com/openthread/openthread/files/520883/SED_9_2_6.zip)

lluwang, 2016-10-11:
[Commissioner_9_2_3.zip](https://github.com/openthread/openthread/files/520963/Commissioner_9_2_3.zip)

lluwang, 2016-10-11:
[Leader_9_2_4.zip](https://github.com/openthread/openthread/files/521177/Leader_9_2_4.zip)

xiaom-GitHub, 2016-10-11:
0754ac7 with Harness V1.1 R26
[Leader_9_2_11.zip](https://github.com/openthread/openthread/files/521631/Leader_9_2_11.zip)
[Router_9_2_11.zip](https://github.com/openthread/openthread/files/521633/Router_9_2_11.zip)
[ED_9_2_11.zip](https://github.com/openthread/openthread/files/521634/ED_9_2_11.zip)
[SED_9_2_11.zip](https://github.com/openthread/openthread/files/521635/SED_9_2_11.zip)

chshu, 2016-10-11:
[Leader_5_8_4.zip](https://github.com/openthread/openthread/files/521834/Leader_5_8_4.zip)

xiaom-GitHub, 2016-10-13:
201a9c1 with Harness V1.1 R26
[Leader_9_2_18.zip](https://github.com/openthread/openthread/files/526012/Leader_9_2_18.zip)
[Router_9_2_18.zip](https://github.com/openthread/openthread/files/526013/Router_9_2_18.zip)
[ED_9_2_18.zip](https://github.com/openthread/openthread/files/526014/ED_9_2_18.zip)

xiaom-GitHub, 2016-10-16:
#821 with Harness V1.1 R26
[Leader_9_2_12.zip](https://github.com/openthread/openthread/files/531570/Leader_9_2_12.zip)
[Router_9_2_12.zip](https://github.com/openthread/openthread/files/531571/Router_9_2_12.zip)

librasungirl, 2016-10-16:
8bb3d4d with Harness R20
[Router_5_1_2.zip](https://github.com/openthread/openthread/files/531681/Router_5_1_2.zip)
[Router_5_3_2.zip](https://github.com/openthread/openthread/files/531680/Router_5_3_2.zip)
[Router_5_3_3.zip](https://github.com/openthread/openthread/files/531682/Router_5_3_3.zip)
[Router_5_3_4.zip](https://github.com/openthread/openthread/files/531683/Router_5_3_4.zip)
[Leader_5_3_7.zip](https://github.com/openthread/openthread/files/531676/Leader_5_3_7.zip)
[Router_5_3_10.zip](https://github.com/openthread/openthread/files/531684/Router_5_3_10.zip)
[Router_5_5_3.zip](https://github.com/openthread/openthread/files/531685/Router_5_5_3.zip)
[Leader_5_5_7.zip](https://github.com/openthread/openthread/files/531677/Leader_5_5_7.zip)
[Router_5_5_7.zip](https://github.com/openthread/openthread/files/531686/Router_5_5_7.zip)
[SED_6_3_1.zip](https://github.com/openthread/openthread/files/531687/SED_6_3_1.zip)

75e7882 with Harness R20
[REED_5_2_4.zip](https://github.com/openthread/openthread/files/531678/REED_5_2_4.zip)
[REED_5_6_7.zip](https://github.com/openthread/openthread/files/531679/REED_5_6_7.zip)

librasungirl, 2016-10-16:
c28042c with Harness R26

[ED_6_1_1.zip](https://github.com/openthread/openthread/files/531695/ED_6_1_1.zip)
[ED_6_1_2.zip](https://github.com/openthread/openthread/files/531696/ED_6_1_2.zip)
[ED_6_1_3.zip](https://github.com/openthread/openthread/files/531697/ED_6_1_3.zip)
[ED_6_1_4.zip](https://github.com/openthread/openthread/files/531700/ED_6_1_4.zip)
[ED_6_1_5.zip](https://github.com/openthread/openthread/files/531698/ED_6_1_5.zip)
[ED_6_1_6.zip](https://github.com/openthread/openthread/files/531699/ED_6_1_6.zip)
[ED_6_2_1.zip](https://github.com/openthread/openthread/files/531701/ED_6_2_1.zip)
[ED_6_2_2.zip](https://github.com/openthread/openthread/files/531702/ED_6_2_2.zip)
[ED_6_3_1.zip](https://github.com/openthread/openthread/files/531703/ED_6_3_1.zip)
[ED_6_4_1.zip](https://github.com/openthread/openthread/files/531704/ED_6_4_1.zip)
[ED_6_4_2.zip](https://github.com/openthread/openthread/files/531706/ED_6_4_2.zip)
[ED_6_5_1.zip](https://github.com/openthread/openthread/files/531707/ED_6_5_1.zip)
[ED_6_5_2.zip](https://github.com/openthread/openthread/files/531705/ED_6_5_2.zip)
[ED_6_5_3.zip](https://github.com/openthread/openthread/files/531708/ED_6_5_3.zip)
[ED_6_6_1.zip](https://github.com/openthread/openthread/files/531709/ED_6_6_1.zip)
[ED_6_6_2.zip](https://github.com/openthread/openthread/files/531714/ED_6_6_2.zip)
[FED_6_1_7.zip](https://github.com/openthread/openthread/files/531710/FED_6_1_7.zip)
[MED_6_3_2.zip](https://github.com/openthread/openthread/files/531723/MED_6_3_2.zip)

[REED_5_2_7.zip](https://github.com/openthread/openthread/files/531722/REED_5_2_7.zip)
[REED_5_5_5.zip](https://github.com/openthread/openthread/files/531725/REED_5_5_5.zip)

[Leader_5_1_1.zip](https://github.com/openthread/openthread/files/531711/Leader_5_1_1.zip)
[Router_5_1_1.zip](https://github.com/openthread/openthread/files/531726/Router_5_1_1.zip)
[Router_5_1_3.zip](https://github.com/openthread/openthread/files/531727/Router_5_1_3.zip)
[Router_5_1_4.zip](https://github.com/openthread/openthread/files/531731/Router_5_1_4.zip)
[Leader_5_1_5.zip](https://github.com/openthread/openthread/files/531713/Leader_5_1_5.zip)
[Router_5_1_6.zip](https://github.com/openthread/openthread/files/531729/Router_5_1_6.zip)
[Router_5_1_7.zip](https://github.com/openthread/openthread/files/531728/Router_5_1_7.zip)
[Router_5_1_8.zip](https://github.com/openthread/openthread/files/531732/Router_5_1_8.zip)
[Router_5_1_9.zip](https://github.com/openthread/openthread/files/531730/Router_5_1_9.zip)
[Router_5_1_10.zip](https://github.com/openthread/openthread/files/531733/Router_5_1_10.zip)
[Router_5_1_11.zip](https://github.com/openthread/openthread/files/531734/Router_5_1_11.zip)
[Router_5_1_12.zip](https://github.com/openthread/openthread/files/531735/Router_5_1_12.zip)
[Leader_5_1_13.zip](https://github.com/openthread/openthread/files/531712/Leader_5_1_13.zip)
[Router_5_1_13.zip](https://github.com/openthread/openthread/files/531736/Router_5_1_13.zip)

[Router_5_2_1.zip](https://github.com/openthread/openthread/files/531768/Router_5_2_1.zip)

[Router_5_3_1.zip](https://github.com/openthread/openthread/files/531737/Router_5_3_1.zip)
[Router_5_3_5.zip](https://github.com/openthread/openthread/files/531738/Router_5_3_5.zip)
[Router_5_3_11.zip](https://github.com/openthread/openthread/files/531739/Router_5_3_11.zip)
[Leader_5_5_1.zip](https://github.com/openthread/openthread/files/531715/Leader_5_5_1.zip)
[Router_5_5_1.zip](https://github.com/openthread/openthread/files/531743/Router_5_5_1.zip)
[Leader_5_5_2.zip](https://github.com/openthread/openthread/files/531716/Leader_5_5_2.zip)
[Router_5_5_2.zip](https://github.com/openthread/openthread/files/531741/Router_5_5_2.zip)
[Leader_5_5_3.zip](https://github.com/openthread/openthread/files/531717/Leader_5_5_3.zip)
[Leader_5_5_4.zip](https://github.com/openthread/openthread/files/531721/Leader_5_5_4.zip)
[Router_5_5_4.zip](https://github.com/openthread/openthread/files/531740/Router_5_5_4.zip)

[Router_5_6_1.zip](https://github.com/openthread/openthread/files/531742/Router_5_6_1.zip)
[Leader_5_6_2.zip](https://github.com/openthread/openthread/files/531719/Leader_5_6_2.zip)
[Router_5_6_3.zip](https://github.com/openthread/openthread/files/531744/Router_5_6_3.zip)
[Leader_5_6_4.zip](https://github.com/openthread/openthread/files/531720/Leader_5_6_4.zip)
[Leader_5_6_5.zip](https://github.com/openthread/openthread/files/531718/Leader_5_6_5.zip)
[Leader_5_6_6.zip](https://github.com/openthread/openthread/files/531724/Leader_5_6_6.zip)
[Router_5_6_9.zip](https://github.com/openthread/openthread/files/531745/Router_5_6_9.zip)
[Router_5_7_1.zip](https://github.com/openthread/openthread/files/531746/Router_5_7_1.zip)
[Router_5_8_2.zip](https://github.com/openthread/openthread/files/531775/Router_5_8_2.zip)
[Router_5_8_3.zip](https://github.com/openthread/openthread/files/531776/Router_5_8_3.zip)

[SED_6_1_1.zip](https://github.com/openthread/openthread/files/531747/SED_6_1_1.zip)
[SED_6_1_2.zip](https://github.com/openthread/openthread/files/531749/SED_6_1_2.zip)
[SED_6_1_3.zip](https://github.com/openthread/openthread/files/531750/SED_6_1_3.zip)
[SED_6_1_4.zip](https://github.com/openthread/openthread/files/531748/SED_6_1_4.zip)
[SED_6_1_5.zip](https://github.com/openthread/openthread/files/531751/SED_6_1_5.zip)
[SED_6_1_6.zip](https://github.com/openthread/openthread/files/531753/SED_6_1_6.zip)
[SED_6_2_1.zip](https://github.com/openthread/openthread/files/531752/SED_6_2_1.zip)
[... cut ...]
[BR_Leader_7_1_1.zip](https://github.com/openthread/openthread/files/531689/BR_Leader_7_1_1.zip)
[BR_Router_7_1_2.zip](https://github.com/openthread/openthread/files/531691/BR_Router_7_1_2.zip)
[BR_Leader_7_1_3.zip](https://github.com/openthread/openthread/files/531690/BR_Leader_7_1_3.zip)
[BR_Router_7_1_4.zip](https://github.com/openthread/openthread/files/531693/BR_Router_7_1_4.zip)
[BR_Router_7_1_5.zip](https://github.com/openthread/openthread/files/531692/BR_Router_7_1_5.zip)
[BR_Router_7_1_8.zip](https://github.com/openthread/openthread/files/531694/BR_Router_7_1_8.zip)

librasungirl, 2016-10-18:
OpenThread Thread 1.1 Certification Status Trend (Oct 16)
[image]

lluwang, 2016-10-18:
[Commissioner_9_2_4.zip](https://github.com/openthread/openthread/files/535770/Commissioner_9_2_4.zip)

xiaom-GitHub, 2016-10-23:
275d22d with Harness R27
[Leader_9_2_12.zip](https://github.com/openthread/openthread/files/546392/Leader_9_2_12.zip)
[Router_9_2_12.zip](https://github.com/openthread/openthread/files/546393/Router_9_2_12.zip)
[MED_9_2_12.zip](https://github.com/openthread/openthread/files/546394/MED_9_2_12.zip)

librasungirl, 2016-10-25:
PR884 with Harness R26
[Leader_5_3_8.zip](https://github.com/openthread/openthread/files/552036/Leader_5_3_8.zip)

xiaom-GitHub, 2016-10-28:
[PR#898](https://github.com/openthread/openthread/pull/898) with Harness R28
[SED_9_2_18.zip](https://github.com/openthread/openthread/files/557934/SED_9_2_18.zip)

xiaom-GitHub, 2016-10-31:
f13a4f8 with Harness R29
[Leader_9_2_9.zip](https://github.com/openthread/openthread/files/561618/Leader_9_2_9.zip)
[Router_9_2_9.zip](https://github.com/openthread/openthread/files/561620/Router_9_2_9.zip)
[Router_9_2_10.zip](https://github.com/openthread/openthread/files/561623/Router_9_2_10.zip)
[ED_9_2_10.zip](https://github.com/openthread/openthread/files/561625/ED_9_2_10.zip)
[SED_9_2_10.zip](https://github.com/openthread/openthread/files/561626/SED_9_2_10.zip)

librasungirl, 2016-11-02:
dc6ad5b-mod with Harness R29
 _(modified to allocate router id by order for test purpose due to known harness issue DEV-1019)_
[Router_5_3_9.zip](https://github.com/openthread/openthread/files/566395/Router_5_3_9.zip)

chshu, 2016-11-03:
15008f0 with Harness R29
[Router_9_2_15.zip](https://github.com/openthread/openthread/files/568597/Router_9_2_15.zip)
[Router_9_2_16.zip](https://github.com/openthread/openthread/files/568596/Router_9_2_16.zip)

chshu, 2016-11-08:
[Leader_5_8_4.zip](https://github.com/openthread/openthread/files/577798/Leader_5_8_4.zip)

xiaom-GitHub, 2016-11-08:
266232e with Harness R30
[ED_9_2_17.zip](https://github.com/openthread/openthread/files/578648/ED_9_2_17.zip)
[SED_9_2_17.zip](https://github.com/openthread/openthread/files/578649/SED_9_2_17.zip)

librasungirl, 2016-11-09:
c4eff62 with Harness R30

[Leader_5_1_1.zip](https://github.com/openthread/openthread/files/581180/Leader_5_1_1.zip)
[Leader_5_1_5.zip](https://github.com/openthread/openthread/files/581181/Leader_5_1_5.zip)
[Leader_5_1_13.zip](https://github.com/openthread/openthread/files/581182/Leader_5_1_13.zip)
[Leader_5_3_7.zip](https://github.com/openthread/openthread/files/581183/Leader_5_3_7.zip)
[Leader_5_5_1.zip](https://github.com/openthread/openthread/files/581184/Leader_5_5_1.zip)
[Leader_5_5_2.zip](https://github.com/openthread/openthread/files/581185/Leader_5_5_2.zip)
[Leader_5_5_7.zip](https://github.com/openthread/openthread/files/581186/Leader_5_5_7.zip)
[Leader_5_6_2.zip](https://github.com/openthread/openthread/files/581187/Leader_5_6_2.zip)
[Leader_5_6_5.zip](https://github.com/openthread/openthread/files/581188/Leader_5_6_5.zip)
[Leader_5_6_6.zip](https://github.com/openthread/openthread/files/581189/Leader_5_6_6.zip)
[Leader_7_1_1.zip](https://github.com/openthread/openthread/files/581190/Leader_7_1_1.zip)

[Router_5_1_1.zip](https://github.com/openthread/openthread/files/581192/Router_5_1_1.zip)
[Router_5_1_3.zip](https://github.com/openthread/openthread/files/581194/Router_5_1_3.zip)
[Router_5_1_4.zip](https://github.com/openthread/openthread/files/581193/Router_5_1_4.zip)
[Router_5_1_6.zip](https://github.com/openthread/openthread/files/581195/Router_5_1_6.zip)
[Router_5_1_8.zip](https://github.com/openthread/openthread/files/581197/Router_5_1_8.zip)
[Router_5_1_9.zip](https://github.com/openthread/openthread/files/581196/Router_5_1_9.zip)
[Router_5_1_10.zip](https://github.com/openthread/openthread/files/581198/Router_5_1_10.zip)
[Router_5_1_11.zip](https://github.com/openthread/openthread/files/581200/Router_5_1_11.zip)
[Router_5_1_12.zip](https://github.com/openthread/openthread/files/581201/Router_5_1_12.zip)
[Router_5_1_13.zip](https://github.com/openthread/openthread/files/581199/Router_5_1_13.zip)
[Router_5_2_1.zip](https://github.com/openthread/openthread/files/581202/Router_5_2_1.zip)
[Router_5_3_1.zip](https://github.com/openthread/openthread/files/581203/Router_5_3_1.zip)
[Router_5_3_5.zip](https://github.com/openthread/openthread/files/581204/Router_5_3_5.zip)
[Router_5_3_10.zip](https://github.com/openthread/openthread/files/581205/Router_5_3_10.zip)
[Router_5_3_11.zip](https://github.com/openthread/openthread/files/581206/Router_5_3_11.zip)
[Router_5_5_1.zip](https://github.com/openthread/openthread/files/581207/Router_5_5_1.zip)
[Router_5_8_2.zip](https://github.com/openthread/openthread/files/581208/Router_5_8_2.zip)
[Router_5_8_3.zip](https://github.com/openthread/openthread/files/581209/Router_5_8_3.zip)

[ED_6_1_1.zip](https://github.com/openthread/openthread/files/581166/ED_6_1_1.zip)
[ED_6_1_2.zip](https://github.com/openthread/openthread/files/581165/ED_6_1_2.zip)
[ED_6_1_4.zip](https://github.com/openthread/openthread/files/581169/ED_6_1_4.zip)
[ED_6_1_5.zip](https://github.com/openthread/openthread/files/581168/ED_6_1_5.zip)
[ED_6_1_6.zip](https://github.com/openthread/openthread/files/581173/ED_6_1_6.zip)
[FED_6_1_7.zip](https://github.com/openthread/openthread/files/581178/FED_6_1_7.zip)
[ED_6_2_2.zip](https://github.com/openthread/openthread/files/581171/ED_6_2_2.zip)
[ED_6_4_1.zip](https://github.com/openthread/openthread/files/581170/ED_6_4_1.zip)
[ED_6_5_2.zip](https://github.com/openthread/openthread/files/581172/ED_6_5_2.zip)
[ED_6_5_3.zip](https://github.com/openthread/openthread/files/581174/ED_6_5_3.zip)
[ED_6_6_1.zip](https://github.com/openthread/openthread/files/581175/ED_6_6_1.zip)
[ED_6_6_2.zip](https://github.com/openthread/openthread/files/581177/ED_6_6_2.zip)

[SED_6_1_1.zip](https://github.com/openthread/openthread/files/581212/SED_6_1_1.zip)
[SED_6_2_1.zip](https://github.com/openthread/openthread/files/581213/SED_6_2_1.zip)
[SED_6_2_2.zip](https://github.com/openthread/openthread/files/581211/SED_6_2_2.zip)
[SED_6_3_1.zip](https://github.com/openthread/openthread/files/581214/SED_6_3_1.zip)
[SED_6_3_2.zip](https://github.com/openthread/openthread/files/581215/SED_6_3_2.zip)

[Joiner_8_1_1.zip](https://github.com/openthread/openthread/files/581179/Joiner_8_1_1.zip)

[Commissioner_8_1_2.zip](https://github.com/openthread/openthread/files/581163/Commissioner_8_1_2.zip)

[Commissioner_8_3_1.zip](https://github.com/openthread/openthread/files/581167/Commissioner_8_3_1.zip)

[Commissioner_9_2_3.zip]
(https://github.com/openthread/openthread/files/581162/Commissioner_9_2_3.zip)
[Leader_9_2_3.zip](https://github.com/openthread/openthread/files/581191/Leader_9_2_3.zip)

[ED_9_2_10.zip](https://github.com/openthread/openthread/files/581176/ED_9_2_10.zip)

[Commissioner_9_2_14.zip](https://github.com/openthread/openthread/files/581164/Commissioner_9_2_14.zip)

[Router_9_2_18.zip](https://github.com/openthread/openthread/files/581210/Router_9_2_18.zip)

xiaom-GitHub, 2016-11-11:
0254cde with Harness R30
[Router_9_2_9.zip](https://github.com/openthread/openthread/files/585546/Router_9_2_9.zip)

xiaom-GitHub, 2016-11-14:
0254cde with Harness R30
[Router_9_2_10.zip](https://github.com/openthread/openthread/files/588290/Router_9_2_10.zip)

xiaom-GitHub, 2016-11-14:
2bc7921 with Harness R31
[Router_9_2_10.zip](https://github.com/openthread/openthread/files/589101/Router_9_2_10.zip)
[ED_9_2_10.zip](https://github.com/openthread/openthread/files/589102/ED_9_2_10.zip)
[SED_9_2_10.zip](https://github.com/openthread/openthread/files/589103/SED_9_2_10.zip)

librasungirl, 2016-11-14:
ca04d4e with Harness R31
[Leader_5_1_1.zip](https://github.com/openthread/openthread/files/589339/Leader_5_1_1.zip)
[Leader_5_1_5.zip](https://github.com/openthread/openthread/files/589342/Leader_5_1_5.zip)
[Leader_5_1_13.zip](https://github.com/openthread/openthread/files/589343/Leader_5_1_13.zip)
[Leader_5_3_6.zip](https://github.com/openthread/openthread/files/589344/Leader_5_3_6.zip)
[Leader_5_3_7.zip](https://github.com/openthread/openthread/files/589346/Leader_5_3_7.zip)
[Leader_5_3_8.zip](https://github.com/openthread/openthread/files/589347/Leader_5_3_8.zip)
[Leader_5_5_1.zip](https://github.com/openthread/openthread/files/589345/Leader_5_5_1.zip)
[Leader_5_5_2.zip](https://github.com/openthread/openthread/files/589348/Leader_5_5_2.zip)
[Leader_5_5_3.zip](https://github.com/openthread/openthread/files/589349/Leader_5_5_3.zip)
[Leader_5_5_4.zip](https://github.com/openthread/openthread/files/589350/Leader_5_5_4.zip)
[Leader_5_5_7.zip](https://github.com/openthread/openthread/files/589351/Leader_5_5_7.zip)
[Leader_5_6_2.zip](https://github.com/openthread/openthread/files/589352/Leader_5_6_2.zip)
[Leader_5_6_4.zip](https://github.com/openthread/openthread/files/589353/Leader_5_6_4.zip)
[Leader_5_6_5.zip](https://github.com/openthread/openthread/files/589354/Leader_5_6_5.zip)
[Leader_5_6_6.zip](https://github.com/openthread/openthread/files/589355/Leader_5_6_6.zip)

[Router_5_1_1.zip](https://github.com/openthread/openthread/files/589359/Router_5_1_1.zip)
[Router_5_1_2.zip](https://github.com/openthread/openthread/files/589360/Router_5_1_2.zip)
[Router_5_1_3.zip](https://github.com/openthread/openthread/files/589361/Router_5_1_3.zip)
[Router_5_1_4.zip](https://github.com/openthread/openthread/files/589362/Router_5_1_4.zip)
[Router_5_1_6.zip](https://github.com/openthread/openthread/files/589363/Router_5_1_6.zip)
[Router_5_1_8.zip](https://github.com/openthread/openthread/files/589367/Router_5_1_8.zip)
[Router_5_1_9.zip](https://github.com/openthread/openthread/files/589366/Router_5_1_9.zip)
[Router_5_1_10.zip](https://github.com/openthread/openthread/files/589365/Router_5_1_10.zip)
[Router_5_1_11.zip](https://github.com/openthread/openthread/files/589368/Router_5_1_11.zip)
[Router_5_1_12.zip](https://github.com/openthread/openthread/files/589369/Router_5_1_12.zip)
[Router_5_1_13.zip](https://github.com/openthread/openthread/files/589370/Router_5_1_13.zip)
[Router_5_2_1.zip](https://github.com/openthread/openthread/files/589372/Router_5_2_1.zip)
[Router_5_3_1.zip](https://github.com/openthread/openthread/files/589371/Router_5_3_1.zip)
[Router_5_3_4.zip](https://github.com/openthread/openthread/files/589375/Router_5_3_4.zip)
[Router_5_3_9.zip](https://github.com/openthread/openthread/files/589376/Router_5_3_9.zip)
[Router_5_3_10.zip](https://github.com/openthread/openthread/files/589377/Router_5_3_10.zip)
[Router_5_3_11.zip](https://github.com/openthread/openthread/files/589378/Router_5_3_11.zip)
[Router_5_5_1.zip](https://github.com/openthread/openthread/files/589379/Router_5_5_1.zip)
[Router_5_5_2.zip](https://github.com/openthread/openthread/files/589380/Router_5_5_2.zip)
[Router_5_5_4.zip](https://github.com/openthread/openthread/files/589381/Router_5_5_4.zip)
[Router_5_6_1.zip](https://github.com/openthread/openthread/files/589383/Router_5_6_1.zip)
[Router_5_6_3.zip](https://github.com/openthread/openthread/files/589385/Router_5_6_3.zip)
[Router_5_8_2.zip](https://github.com/openthread/openthread/files/589388/Router_5_8_2.zip)
[Router_5_8_3.zip](https://github.com/openthread/openthread/files/589389/Router_5_8_3.zip)

[ED_6_1_1.zip](https://github.com/openthread/openthread/files/589324/ED_6_1_1.zip)
[ED_6_1_2.zip](https://github.com/openthread/openthread/files/589325/ED_6_1_2.zip)
[ED_6_1_3.zip](https://github.com/openthread/openthread/files/589327/ED_6_1_3.zip)
[ED_6_1_4.zip](https://github.com/openthread/openthread/files/589328/ED_6_1_4.zip)
[ED_6_1_5.zip](https://github.com/openthread/openthread/files/589326/ED_6_1_5.zip)
[ED_6_1_6.zip](https://github.com/openthread/openthread/files/589330/ED_6_1_6.zip)
[ED_6_2_1.zip](https://github.com/openthread/openthread/files/589329/ED_6_2_1.zip)
[ED_6_2_2.zip](https://github.com/openthread/openthread/files/589331/ED_6_2_2.zip)
[ED_6_3_1.zip](https://github.com/openthread/openthread/files/589332/ED_6_3_1.zip)
[ED_6_4_1.zip](https://github.com/openthread/openthread/files/589335/ED_6_4_1.zip)
[ED_6_4_2.zip](https://github.com/openthread/openthread/files/589333/ED_6_4_2.zip)
[ED_6_5_1.zip](https://github.com/openthread/openthread/files/589334/ED_6_5_1.zip)
[ED_6_5_2.zip](https://github.com/openthread/openthread/files/589336/ED_6_5_2.zip)
[ED_6_5_3.zip](https://github.com/openthread/openthread/files/589337/ED_6_5_3.zip)
[ED_6_6_1.zip](https://github.com/openthread/openthread/files/589338/ED_6_6_1.zip)
[ED_6_6_2.zip](https://github.com/openthread/openthread/files/589340/ED_6_6_2.zip)
[FED_6_1_7.zip](https://github.com/openthread/openthread/files/589341/FED_6_1_7.zip)

[SED_6_1_1.zip](https://github.com/openthread/openthread/files/589393/SED_6_1_1.zip)
[SED_6_1_2.zip](https://github.com/openthread/openthread/files/589395/SED_6_1_2.zip)
[SED_6_1_3.zip](https://github.com/openthread/openthread/files/589396/SED_6_1_3.zip)
[SED_6_1_4.zip](https://github.com/openthread/openthread/files/589397/SED_6_1_4.zip)
[SED_6_1_5.zip](https://github.com/openthread/openthread/files/589398/SED_6_1_5.zip)
[... cut ...]
[Leader_7_1_1.zip](https://github.com/openthread/openthread/files/589356/Leader_7_1_1.zip)
[Router_7_1_2.zip](https://github.com/openthread/openthread/files/589390/Router_7_1_2.zip)
[Leader_7_1_3.zip](https://github.com/openthread/openthread/files/589357/Leader_7_1_3.zip)
[Router_7_1_4.zip](https://github.com/openthread/openthread/files/589391/Router_7_1_4.zip)
[Router_7_1_5.zip](https://github.com/openthread/openthread/files/589392/Router_7_1_5.zip)
[Router_7_1_8.zip](https://github.com/openthread/openthread/files/589394/Router_7_1_8.zip)

chshu, 2016-11-17:
fc73bc3 with Harness R31
[Leader_5_8_4.zip](https://github.com/openthread/openthread/files/596658/Leader_5_8_4.zip)
[Commissioner_8_1_1.zip](https://github.com/openthread/openthread/files/596659/Commissioner_8_1_1.zip)
[Commissioner_8_1_2.zip](https://github.com/openthread/openthread/files/596660/Commissioner_8_1_2.zip)
[Commissioner_8_1_6.zip](https://github.com/openthread/openthread/files/596661/Commissioner_8_1_6.zip)
[Commissioner_8_2_1.zip](https://github.com/openthread/openthread/files/596662/Commissioner_8_2_1.zip)
[Commissioner_8_2_2.zip](https://github.com/openthread/openthread/files/596665/Commissioner_8_2_2.zip)
[Commissioner_8_2_5.zip](https://github.com/openthread/openthread/files/596664/Commissioner_8_2_5.zip)
[Commissioner_8_3_1.zip](https://github.com/openthread/openthread/files/596663/Commissioner_8_3_1.zip)
[Joiner_8_1_1.zip](https://github.com/openthread/openthread/files/596666/Joiner_8_1_1.zip)
[Joiner_8_1_6.zip](https://github.com/openthread/openthread/files/596667/Joiner_8_1_6.zip)
[Router_8_2_1.zip](https://github.com/openthread/openthread/files/596668/Router_8_2_1.zip)
[Router_8_2_2.zip](https://github.com/openthread/openthread/files/596670/Router_8_2_2.zip)
[Router_8_2_5.zip](https://github.com/openthread/openthread/files/596669/Router_8_2_5.zip)
[ED_9_2_8.zip](https://github.com/openthread/openthread/files/596671/ED_9_2_8.zip)
[Router_9_2_8.zip](https://github.com/openthread/openthread/files/596672/Router_9_2_8.zip)

librasungirl, 2016-11-18:
[PR#991](https://github.com/openthread/openthread/pull/991) with Test Harness V1.1 Beta v1.0
[REED_5_2_5.zip](https://github.com/openthread/openthread/files/600054/REED_5_2_5.zip)

librasungirl, 2016-11-21:
9de7f12 with Harness V1.1 Beta v1.0
[REED_5_2_5.zip](https://github.com/openthread/openthread/files/602539/REED_5_2_5.zip)

[27 more replies]

closed 2018-01-29 by jwhui