openthread/openthread #367: openthread compilation error

truptikumbhar91, 2016-08-12:
I am trying to compile the openthread source code for cc2538 platform.I am following steps given on below link: https://github.com/openthread/openthread/blob/master/examples/apps/cli/README.md
$ cd <path-to-openthread>
$ ./bootstrap
$ make -f examples/Makefile-cc2538
But I am not able to compile it. I am getting following error while compilation
checking for suffix of object files... configure: error: in `/home/trupti/open_thread/openthread/build/cortex-m3-arm-none-eabi':
configure: error: cannot compute suffix of object files: cannot compile
See `config.log' for more details
make: *** [build/cortex-m3-arm-none-eabi/config.status] Error 1

I am using Ubuntu 12.04 for compilation

jwhui, 2016-08-12:
This seems to be a common error and is covered in the [GCC FAQ](https://gcc.gnu.org/wiki/FAQ#Configuration_fails_with_.27.27configure:_error:_cannot_compute_suffix_of_object_files:_cannot_compile.27.27._What_is_the_problem.3F).

The cc2538 build is passing on [Travis CI](https://travis-ci.org/openthread/openthread/).  You can reference the [Job Log](https://travis-ci.org/openthread/openthread/jobs/151845236) to see how the build environment is set up on a clean Ubuntu VM.

closed 2016-08-12 by jwhui

truptikumbhar91, 2016-08-16:
Means should I need to install  MPFR, MPC and GMP ?
Can u please provide me proper link to install above packets.

truptikumbhar91, 2016-08-22:
@jwhui : Installed  MPFR, MPC and GMP libraries using following commands
$sudo apt-get install libgmp-dev
$sudo apt-get install libmpfr-dev
$sudo apt-get install libmpc-dev 
But still getting same error and config.log is also same.