openthread/openthread #369: spinel `whitelist clear` returns failed requirement

turon, 2016-08-12:
Clearing the whitelist is presumed to be done by sending a SPINEL_CMD_PROP_VALUE_SET of SPINEL_PROP_MAC_WHITELIST to an empty value.  That however results in the following failed requirement:

```
spinel-cli > whitelist clear
TX Raw: (8) ['7E', '80', '04', '80', '26', '29', '7A', '7E']
spinel.c:340: Requirement Failed (data_len >= sizeof(spinel_eui64_t))
RX Pay: (6) ['80', '06', '00', '09', 'A8', '9A'] 
PROP_VALUE_IS: LAST_STATUS = 9
Done
```

Breaking down what was sent:
'80', = HDR
'04', = CMD_PROP_SET
'80', '26', = 0x1300 PROP_WHITELIST in EXI integer coding
[ no value follows, just the FCS16 ]

darconeous, 2016-08-13:
Having a look.

darconeous, 2016-08-13:
I think I have an understanding of what is going on. Hopefully I'll have a pull request shortly.

darconeous, 2016-08-13:
Oh! `0x04` is `CMD_PROP_VALUE_INSERT`, not `CMD_PROP_VALUE_SET`!

`CMD_PROP_VALUE_SET` is `0x03`. :)

darconeous, 2016-08-13:
Gosh I thought I was going crazy there for a bit. :P

closed 2016-08-15 by jwhui