openthread/openthread #383: Thread1.1 5.3.7 Duplicate Address Detection Analysis

xiaom-GitHub, 2016-08-16:
5.3.7 fails due to IPv6 source address is not the Rloc of DUT as a leader.(::ff:fe00:0)
Test Script will check if the source address of Address Error Coap packet is fd00::ff:fe00:0 as below:
"stepResults.update(ScriptHelper.validate_CoAP_Address_Error_Packet(mleAdvtPacket, strKey="3",checkAllTLVs=True,Ipv6_Src_Address_Check ="::ff:fe00:0"))"

Currently we use random router id when allocating, even for leader router id. Hence, above criteria fails.

Question:
May I know the reason why we choose random router id instead of an order one?

Enhancement:
Thread1.1 defines the address error message as CON instead of NON. hence, we need to response it when receiving a/ae.

CoAP Request URI:
NON POST coap://[<peer address>]:MM/a/ae

Transaction Pattern:
Ntf/Qry/Ans_CON

jwhui, 2016-08-16:
Part of the reason we use random router ID assignment is to increase the likelihood that routers can keep the router ID when partitions merge.  If every leader uses the same deterministic method for allocating router IDs, then partitions that merge will always have router ID conflicts that need to be resolved.  If the spec says anything about router ID assignment, please let me know.

xiaom-GitHub, 2016-08-16:
@jwhui Thanks much for this clarification, that indeed makes sense. Go through the spec again, Router ID assignment section does not require that allocating router ID starting from zero(child ID must be 0 reserved for router). Please let me file a JIRA to check about this verification restriction. Thanks.

PR #400 (2016-08-17): update address error notification message with Confirmable flag.
Resolves #383.
closed 2016-08-17 by jwhui