openthread/openthread #385: border router implementation

Praveen-Kumar-K, 2016-08-16:
Hi All,
I have seen thread cli example it is working fine. I want to implement thread network with my Native PC or hardware boards (cc2538). My question is:
1. Is Leader will do the functionality of border router, by adding special functionality(What would be the requirements) or separate Border router (Hardware) require?
2. Is there any specific hardware to implement Border router in openthread? if so let me know please.
can any one share the procedure and details for implementing thread network and send message between child, router and Leader.

Thanks,
Praveen

jwhui, 2016-08-16:
You can use [wpantund](https://github.com/openthread/wpantund) to expose a Thread network via a standard IPv6 network interface on Unix-like operating systems.

closed 2016-08-16 by jwhui

Praveen-Kumar-K, 2016-08-16:
Thanks @jwhui,
how i can implement on Hardware, what hardware (Border router) should i require and how can i do it. let me Know please.

Thanks,
Praveen

jwhui, 2016-08-16:
The [wpantund wiki](https://github.com/openthread/wpantund/blob/master/INSTALL.md#wpantund-installation-guide) has installation instructions for Ubuntu and Mac OS X.