openthread/openthread #407: spinel: documentation issue with local routes property

turon, 2016-08-17:
PROP_THREAD_LOCAL_ROUTES (route add/remove) looks to have the same arguments as PROP_THREAD_ON_MESH_NETS (prefix add/remove), but the stable parameter should probably be               prf instead: Default Router Preference, which may be: 'high', 'med', or 'low'.

Also, for PROP_THREAD_ON_MESH_NETS, the stable flag should probably just be part of the general flags parameter and follow the Thread specification definition, otherwise the API has a  conflicting redundant way of setting it.

```
D.4.12. PROP 91: PROP_THREAD_LOCAL_ROUTES

Type: Read-Write
Packed-Encoding: A(T(6CbC))
Data per item is:

6: IPv6 Prefix
C: Prefix length, in bits
b: Stable flag
C: Other flags
```

turon, 2016-08-17:
per RQ: stable flag is broken out specifically because it is determined by the encapsulating TLV, not the flags.

Documentation should provide more detail or spec reference for the encoding of "Other flags".

abtink, 2017-06-09:
PR https://github.com/openthread/openthread/pull/1886 addresses this issue.

PR #1886 (2017-06-09): NcpBase: Update getter for `OFF_MESH_ROUTES` to include `NextHopIsThisDevice` field
The spinel header file and spinel documentation are also updated
accordingly.

Also resolves #407.
closed 2017-06-09 by jwhui