openthread/openthread #413: one question about router id allocation

librasungirl, 2016-08-18:
The scenario is with one thread network with 1 Leader, 1 Router:
Leader release Router's router id via cli, then Router will try to reattach again when recognizing its router ID has been removed.  Rloc Tlv would be included in the Router's address solicit message trying to get same router id, should the Leader accept and assign same router id to it or should it reject and allocate a different router id to it? which is preferred?

jwhui, 2016-08-18:
I guess it depends on the situation.  Generally speaking, routers should try to keep their existing Router ID to minimize disruption to any children they may have attached to them.  However, if there are other reasons (e.g. testing), I don't see any issue with adding some functionality to the leader that forces the router to have a different Router ID.

librasungirl, 2016-08-19:
Got it. Thanks.

closed 2016-08-19 by librasungirl