openthread/openthread #443: analysis: Thread1.1 6.3.2 Network data update

xiaom-GitHub, 2016-08-22:
6.3.2 test case fails when checking if CIDs corresponding to all configured on-mesh prefix are present in Address Registration TLV in Child Update Request message.

Currently for sleepy end device, after it obtains the new network data(i.e. on-mesh prefix propagated from border router), it will send two child update request messages continuously to its parent. The first child update message occurs in `HandleNetworkDataUpdate` after sleepy end device set the new network data; and the second one occurs in `HandleNetifStateChanged` after a new IPv6 unicast address added on thread interface.

Harness will locate the first child update message after leader multicast a Data Response with new configured on-mesh prefix(i.e. 2001::/64). However, for the first child update message, Address Registration TLV still does not include the new configured IPv6 unicast address, hence, it does not include CID neither. The screenshot as below:
[image]

Notice that Thread 1.1 spec 4.7.3 states:
"A child update Request may be sent by a child in the following circumstances:
 To modify its mode, timeout duration, and/or address registration"

@jwhui @librasungirl Is it essential to send the first child update request in `HandleNetworkDataUpdate` for a sleepy end device? or if there is another scenario that require to send the child update request in `HandleNetworkDataUpdate`, please let me know. Thanks.

jwhui, 2016-08-22:
Thanks for this analysis.  Triggering a Child Update Request message in `HandleNetworkDataUpdate()` was added as part of #404.  It is necessary to notify its parent that it had received an update to the Network Data.

I think the fix is to handle the network data update event as part of `HandleNetifStateChanged()` so that the two updates can be coalesced into one event.  I can submit a PR soon.

librasungirl, 2016-08-24:
Hi @jwhui , I guess #444 is meant to trigger only one SendChildUpdateRequest() due to network data update no matter it includes prefix update or not. Please ignore the following if it is not correct.
 I think #444 just postpone SendChildUpdateRequest() triggered by OT_THREAD_NETDATA_UPDATED because SetStateChangedFlags() would post a task for it. Still there would be  SendChildUpdateReqeust() triggered by (OT_IP6_ADDRESS_ADDED | OT_IP6_ADDRESS_REMOVED) when processing network data if it includes prefix updates.

That is, there would still be two separate SendChildUpdateRequest() triggered by OT_THREAD_NETDATA_UPDATED and then (OT_IP6_ADDRESS_ADDED | OT_IP6_ADDRESS_REMOVED) when network data includes prefix update.
Due to task mechanism, there is chance that when sending first Child Update Request, the addresses have been updated so that the address register tlv would include all addresses should be registered. But that might not be guaranteed.

jwhui, 2016-08-24:
@librasungirl, `SetStateChangedFlags()` should post a task to delay the handler until after the current task is complete.  If both the IPv6 addresses and Network Data are updated in the same task, then `SendChildUpdateRequest()` should only be called once.

However, if the IPv6 addresses and Network Data are updated in separate tasks, then `SendChildUpdateRequest()` may be called twice.  Do you see a case where this can happen?

librasungirl, 2016-08-25:
IPv6 addresses and Network Data would not be in the same task because IPv6 addresses are updated when handling tasks for network data update `Interpreter::HandleNetifStateChanged`
So we would always see two `SendChildUpdateRequest()`as in the below pcap.

[Test_6_3_2_Network_Update_(End_Device[MED])_2016_08_25-10_20_08_(Ch-16).pcapng.zip](https://github.com/openthread/openthread/files/436067/Test_6_3_2_Network_Update_.End_Device.MED._2016_08_25-10_20_08_.Ch-16.pcapng.zip)

jwhui, 2016-08-26:
Good catch!  I just submitted #476 to fix this.

PR #444 (2016-08-23): Handle Network Data updates via netif handler to coalesce message transmissions.
Resolves #443.
PR #476 (2016-08-26): Only send Child Update Request message when no more state changes are pending.
Resolves #443.
closed 2016-08-26 by jwhui