openthread/openthread #470: Device can't rejoin network after hardware reset

chshu, 2016-08-25:
In a formed OpenThread network, if one device run into a hardware reset (all the parameters will be reset), the device can not rejoin the network again due to `Frame Counter` mismatch, then it will become a leader and form a new partition, and will go through a partition merging process.

Not sure if it is as design, or a potential issue needs to be fixed? And I failed to find any corresponding description in specification talking about this case (rejoin after hardware reset). Appreciate for any feedbacks, thanks.

jwhui, 2016-08-25:
Thread 1.1 requires a device to generate a new random MAC Extended Address when it is commissioned.  As a result, this should not be an issue.  While MAC Extended Address collisions may occur, the probability is exceedingly low.

chshu, 2016-08-26:
Got it, Thanks @jwhui for the answer. 

So currently we have the same MAC Extended Address after hardware reset which causes rejoin failure, but after enabling `Commissioning` feature, the device will generate a different MAC Extended Address and could rejoin the network successfully. Please correct me if I have any misunderstanding, thanks.

chshu, 2016-08-26:
Maybe I have had some misunderstanding...
According to the spec, device needs to generate a new random MAC Extended Address whether enabling {{Commissioning}} feature or not.

It seems that it is not an issue of OpenThread stack, but an issue of `otPlatRandomGet()` implementation in platform layer, both CC2538 and posix platform provide a pseudo-random number generator, and generate the same MAC Extended Address after each reset.

jwhui, 2016-08-26:
@chshu, you are correct that the MAC Extended Address must always be randomly generated.  This needs to be done whether you are commissioned in-band using Thread Commissioning or out-of-band by some other means.

The current examples for `otPlatRandomGet()` do not satisfy the requirements for generating MAC Extended Addresses.  Ideally, the MAC Extended Address will be generated using a true random source.

turon, 2016-08-26:
Here is a code snippet that could be used to seed the pseudo-random function from the true-random ADC:

```
int cc2538_random_read(uint8_t *outEntropy, size_t inSize)
{
    uint32_t            randword; 
    unsigned char       *pbuf           = outEntropy;
    const size_t        req_words       = inSize >> 2;
    const unsigned char *pbuf_end       = outEntropy + inSize;
    const unsigned char *preq_words_end = 
      (unsigned char *)(((uint32_t *)outEntropy) + req_words);
    ptrdiff_t           remainder_bytes = pbuf_end - preq_words_end;

    for (pbuf = outEntropy; pbuf < preq_words_end; pbuf += 4)
    {
        HWREG(SOC_ADC_ADCCON1) |= 0x00000004;
        randword = HWREG(SOC_ADC_RNDL) | (HWREG(SOC_ADC_RNDH) << 8);
        memcpy(pbuf, &randword, 4);
    }

    if (remainder_bytes)
    {
        HWREG(SOC_ADC_ADCCON1) |= 0x00000004;
        randword = HWREG(SOC_ADC_RNDL) | (HWREG(SOC_ADC_RNDH) << 8);
        memcpy(pbuf, &randword, remainder_bytes);
    }

    return 0;
}
```

jwhui, 2016-08-26:
@turon, that does not work.  The SOC_ADC_RNDL and SOC_ADC_RNDH registers implement a pseudo-random sequence.  For the CC2538, the suggested method is to utilize the IF_ADC in the RF receive path.

turon, 2016-08-27:
Yes, you're right.  It seems like we have a todo to properly fix the random seeding on cc2538 to follow the advised method in the chip reference manual.

jwhui, 2016-08-27:
@turon, that effort is already tracked as part of  #354.

closed 2016-09-09 by chshu