openthread/openthread #486: Thread Certification results as hosted HTML

beriberikix, 2016-08-29:
Currently we host [Thread Certification results](https://github.com/openthread/openthread/wiki/Thread-Certification#results-summary) as PDFs in ZIPs. It would be faster to check results if they were hosted as HTML. We can leverage the [new feature for publishing docs](https://github.com/blog/2233-publish-your-project-documentation-with-github-pages).

jwhui, 2016-08-29:
I agree it would be nice to have the results in HTML format.  However, the Thread Certification Harness produced by the Thread Group only outputs results in PDF format.

beriberikix, 2016-08-29:
That's what I figured. Let's put it on the backburner for now (P4? Another label?) Maybe it is something we can ask for in the future.

turon, 2016-08-29:
Does the publishing docs feature support hosing PDF files via HTTP?

_____________________________
Martin Turon  |  Nest Labs

jwhui, 2016-08-29:
The GitHub feature will host anything you can commit to git.  However, linking a PDF file is pretty much what we have now on the wiki.

darconeous, 2016-09-15:
There is always <https://rawgit.com>.

jwhui, 2017-10-02:
Closing stale issue.

closed 2017-10-02 by jwhui