openthread/openthread #492: ☂ Spinel NCP Protocol Standardization

darconeous, 2016-08-30:
This issue is tracking the efforts, issues, and discussion regarding the documentation and standardization of the "Spinel" NCP protocol. This is the place for general discussion about this effort.

Please relate all Spinel standardization issues to this issue by adding the [☂ Spinel Standardization Tag](https://github.com/openthread/openthread/labels/%E2%98%82%20Spinel%20Standardization) to them.

darconeous, 2016-08-30:
Will likely use the process outlined in [rfc7328](https://tools.ietf.org/html/rfc7328.html) for the official protocol documentation.

darconeous, 2016-09-02:
I'm considering using [mmark](https://github.com/miekg/mmark) instead of [pandoc2rfc](https://github.com/miekg/pandoc2rfc) (as described in RFC 7328), since the author of pandoc2rfc states the following in the [pandoc2rfc README](https://github.com/miekg/pandoc2rfc/blob/master/README.md):

> Note that I consider [mmark](https://github.com/miekg/mmark) the logical successor of Pandoc2rfc. Pandoc2rfc can't output XML2RFC v3 XML, so when the IETF switches over to that format, you'll need something else. Mmark can output v3 (and v2 for that matter).

darconeous, 2016-09-02:
I've successfully moved all of the existing documentation over to mmark. You can get a taste of the result [here](https://gist.github.com/darconeous/262e659d1159f02261031733ba37da22#file-draft-spinel-protocol-bis-txt). I've reorganized the document to help aid clarity. I've also added a test-vector section, but it is currently empty at the moment.

darconeous, 2016-09-07:
PR #534 is for integrating the documentation refactoring into OpenThread's master branch. Once that is in place, it will be easier for people like @beriberikix to help with document editing.

There is still a lot of work to do, but I think things are moving along at a reasonable pace.

beriberikix, 2016-09-08:
👍 /cc @ackermanc

darconeous, 2018-07-09:
Closing. See https://github.com/openthread/spinel-spec

closed 2018-07-09 by darconeous