openthread/openthread #517: spinel-cli: fix failing tests

jwhui, 2016-09-02:
spinel-cli was merged in #364, however multiple tests still fail.  This issue tracks the effort to get all tests passing.

jwhui, 2016-09-06:
I'm seeing spurious pass/fail results on Cert_5_1_07 with ncp-sim ([build example](https://travis-ci.org/openthread/openthread/jobs/157433184)).  I'm going to submit a another PR that moves that test to the SKIP list for now.

jwhui, 2016-09-13:
I'm still seeing spurious failures ([example](https://travis-ci.org/jwhui/openthread/jobs/159651444)).

turon, 2016-09-13:
@jwhui @darconeous #585 leverages the new ot-ncp clean exit to hopefully solve the spurious eof failures.

jwhui, 2016-09-14:
Still seeing spurious failures ([example](https://travis-ci.org/openthread/openthread/jobs/159939839))...

turon, 2016-09-14:
Thanks for letting me know.  Root cause for that seems to be this:

```
ot-ncp: /home/travis/build/openthread/openthread/examples/../src/core/common/message.cpp:201: ThreadError Thread::Message::MoveOffset(int): Assertion `GetOffset() + aDelta <= GetLength()' failed.
Aborted (core dumped)
```

PR #669 (2016-09-23): spinel-cli: Add ipaddr removal to prefix change handler.
Fixes Cert_5_6_06 and Cert_5_6_08.  Fixes #517.
Collapse fcs16 into Hdlc class.
Add and correct subcommand completion handlers.
Move prefix handler to separate worker thread and task queue.
Channelize blocking spinel property change handlers to allow parallel
reuse across multiple threads.
commit a853ffc7 (2016-09-23): spinel-cli: Add ipaddr removal to prefix change handler. (#669)

Fixes Cert_5_6_06 and Cert_5_6_08.  Fixes #517.
Collapse fcs16 into Hdlc class.
Add and correct subcommand completion handlers.
Move prefix handler to separate worker thread and task queue.
Channelize blocking spinel property change handlers to allow parallel
reuse across multiple threads.
closed 2016-09-23 by jwhui