openthread/openthread #537: Spinel: ON_MESH format inconsistent

turon, 2016-09-08:
The documentation for the ON_MESH (prefix) parameter is `A(T(6CbC))`, but it seems the GET handler returns `A(T(6CbCb))`, where the final parameters is the isLocal flag.  INSERT and REMOVE continue to expect no isLocal flag be passed.

```
#### D.4.11. PROP 90: `PROP_THREAD_ON_MESH_NETS`
* Type: Read-Write
* Packed-Encoding: `A(T(6CbCb))`

Data per item is:

* `6`: IPv6 Prefix
* `C`: Prefix length, in bits
* `b`: Stable flag
* `C`: Other flags
* `b`: Is Local flag   <=== NEW
```

turon, 2016-09-09:
@darconeous do you agree with this issue?

jwhui, 2016-09-09:
@turon, this is already being addressed.  Please see the referenced PR above.

turon, 2016-09-09:
@jwhui Ok, thanks for pointing that out.  I just want to make sure the fix goes beyond the documentation issue and addresses INSERT and REMOVE not expecting the new parameter.

jwhui, 2016-09-09:
@turon, yes those questions are also raised in the PR.

darconeous, 2016-09-15:
INSERT and REMOVE shouldn't require the islocal flag to be passed in our implementation. It should ignore the flag---but that's really an implementation detail.

I've updated the official documentation. Let me know if you think the update is adequate.

closed 2016-09-16 by darconeous