openthread/openthread #570: A Number of Build Warnings on Windows

nibanks, 2016-09-12:
Many of the build warnings have been fixed in the last few months, but a number are still there. I recently got appveyor building on my branches, and you can see the list of warnings from the latest build:

https://ci.appveyor.com/project/nibanks/openthread/build/messages

Most of the build warnings are "conversion from .. to .., possible loss of data"

jwhui, 2016-09-12:
I agree we should fix these build warnings.  Interesting that gcc and clang do not pick up these warnings.  Are you willing to submit a PR that addresses these warnings?

nibanks, 2016-09-12:
Sure. I just figured I'd open an issue now that I had something concrete to point it to.

jwhui, 2016-09-13:
Fixed in #577.

closed 2016-09-13 by jwhui